MarlinFirmware/Marlin #12150: [FR]  Filaments list presets for Extruder steps calibration

ZhuJo, 2018-10-19:
As there is a constant need to calibrate extruder steps after each filament change, it would be great to have ability to create a custom filament list with predefined extruder steps.

User could then choose a filament from list he has created for a specific extruder calibration right after filament change.

A list could be created by sending Gcode command for easier naming directly on computer.

Steps to implementation:
1.) Calibrate extruder steps and save by M92 Exxx;
2.) Save current filament steps via gcode and name an entry for these settings ie. M145 S[index] Exxx (steps calibrated) PRUSA_PETG_BLK (for naming)
3.) M500 to save

Then, when user change the filament, he should be able to go to Filaments menu and select predefined filament he created in steps above.

thinkyhead, 2018-10-26:
Sounds like a job for the slicer's filament profile. Seems like just more noise to burden the firmware with this.

ZhuJo, 2018-10-26:
Might be. Now the idea is that it would be in the firmware, while implementation in slicer engine of your choice might never come.

AnHardt, 2018-10-26:
"Extrusion multiplier" is in Slicer (filament settings) since ages.

ZhuJo, 2018-10-26:
Extrusion multiplier is something else as precisely defined Extruder steps. Moreover, not everybody is using Slicer, personally I hate it.

boelle, 2019-03-13:
Then maybe do the same in cura? or whatever slicer you want

boelle, 2019-06-22:
missing label

closed 2019-07-11 by shitcreek