MarlinFirmware/Marlin #12151: FWRETRACT and SINGLENOZZLE TOOL CHANGE coexistence ' not an issue'

studiodyne, 2018-10-19:
@thinkyhead 
Hi , it's not an issue , just a suggestion
For all that use ' FWRETRACT ' , we use the Long Retract for tool changing
But now , tool changing is implemented
Is it possible to simplify the code to have only one possibility and no coexistence because 
-Tool change need to be set to ZERO , to use our actual flavor , because long retract + tool change = double long retract
- If we set FWRETRACT swap to zero , or TOOLCHANGE to zero , we have some sram eated and not used , because of float variables and menu implementations

Possible to have a codependent system ?
Like this 
#define  TOOLCHANGE_USE_FWRETRACT_SWAP_SETTINGS 'no toolchange menus, swapping statuses are updated when toolchanging'
or
#define  DISABLE_FWRETRACT_SWAPPING_BY_USING_TOOLCHANGE ' no fwretract long retract menus, no need of new flavor in our slicer tool change tab'
or
#define DISABLE_TOOLCHANGE_SYSTEM_WHEN_FWRETRACT_ENABLED 'no need to change or flavor and no toolchange menus '

or the better for me 
#define INTELLIGENT_TOOLCHANGE_FWRETRACT_COEXISTENCE 
This option use the FWretract setting , toolchange menu deleted , and a smart filter that avoid a bad behavior , like this : If swapped , the toolchange process don't unload , just apply new tool and load it , update 'not swapped status ' to the actual. The tool change make a verification of the swapping statuses before changing tool and update all swap statuses

Just an idea for thinkyhead

InsanityAutomation, 2018-10-19:
Im working on consolidating some more stuff in tool change. In that, there is a specific enable for filament swap on tool change and im expanding it to all tool change types. From this, it should be a matter of not enabling the dedicated filament swap if you're using the fwretract settings to accommodate.

studiodyne, 2018-10-19:
i have rewritten my suggestion , please read once again 
i'm talking about implementing the LOADING STATUS OF THE EXTRUDERS , and will be simplified

InsanityAutomation, 2018-10-19:
There is branches in tool change to allow functionality even when fwretract is disabled. Hold off a few days and take a fresh look when I get a bit farther on what im working on, and it may  be better to see whats going on.

InsanityAutomation, 2018-10-19:
I just updated my current working branch to pr #12137 so maybe you can get a picture of where im going.

closed 2018-10-26 by thinkyhead