MarlinFirmware/Marlin #12154: [2.0.x] TMC2100 config clarification

ModMike, 2018-10-19:
When configuring a TMC2100, what settings does it take on?  If I am using it in stand alone mode should I just not define it in XYZ?

ModMike, 2018-10-19:
Clarification:

When running these do I need to manually define:

```
//#define MINIMUM_STEPPER_DIR_DELAY 650
//#define MINIMUM_STEPPER_PULSE 2
//#define MAXIMUM_STEPPER_RATE 250000
```
or is it taken care of somewhere else?  I am afraid that if I set these global settings my A4988 extruder drivers will be unhappy.

kAdonis, 2018-10-19:
You dont need to manually define the timing in Configuration_adv.h
just set the driver Types in Configuration.h
As I understand, the "slowest" driver determines the global settings
by the way: the TMC2100 has no SPI or serial interface, so it is always in standalone mode

closed 2018-10-20 by ModMike