MarlinFirmware/Marlin #12156: [2.0.x] Z axis only moves upwards after homing

dahuafschmied, 2018-10-19:
### Description
Hi
I have an Anycubic i3 Mega with 8bit board, 5x TMC2208 standalone and REPRAP_DISCOUNT_SMART_CONTROLLER.
I am using the lastest Version of Marlin 2 from Github.
Here is my Repository: https://github.com/dahuafschmied/Marlin-for-Anycubic-i3-Mega/tree/bugfix-2.0.x
i have changed the config, config_adv, trigorilla1.4 pins and configuration_store (for #12112 fix)
here are the changes: https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...dahuafschmied:bugfix-2.0.x

The bug:
I can only move the z axis downward (nozzle to bed) bevore homing.
After Homing the z axis is only moveable upwards.

### Steps to Reproduce
compile my config.
move the z axis up and down
home the z axis
only up is possible

#### Additional Information
Video:
https://www.facebook.com/flatti/videos/10205276294664664/

Marlin 1.1.9 -> OK (up and down possible)
Disabled  #define Z_DUAL_ENDSTOPS  -> OK ( up and down possible, but this is needed, printer has 2 endstops and 2 stepper for Z)

checked the endstops with M119. -> they are OK
checked the endstops with M43 E0 -> they are OK

is the z axis moved downwards after homing the steppers are not spinning but the position is decremented on the display.

Thank you for your help

GMagician, 2018-10-19:
You don't have a updated bugfix 2.0.version, it seems you miss #11900

GMagician, 2018-10-19:
@dahuafschmied Instead of enabling interrupt feature you may you may try to remove line 281 (`#if ENABLED(ENDSTOP_INTERRUPTS_FEATURE)`) in endstops.cpp (and its `#endIf` counterpart).
If it works I'll post a PR

dahuafschmied, 2018-10-19:
@GMagician = hero! :-)
i can confirm your patch is fine. 

interrupt: my Y_MIN_PIN and Z2_MIN_PIN is not interrupt-capable

PR #12158 (2018-10-19): [2.0.x] Fix endstop when no interrupt
#11900 is needed also when interrupt endstop are disabled

Should fix #12156
closed 2018-10-19 by dahuafschmied