MarlinFirmware/Marlin #12159: [2.0.X] Missing NeoPixel Implementation for LPC176X (Azteeg X5 GT)

andyman198, 2018-10-19:
### Description
The neopixel library isn't implemented for the LPC176X, in my case the Azteeg X5 GT. I have found a NeoPixel library that supports the LPC1768 here: https://os.mbed.com/users/JacobBramley/code/PixelArray/
I have not tested using this instead of the standard neopixel library though.

### Steps to Reproduce
Enable the NeoPixel function for an LPC176X based board, firmware will not compile

gloomyandy, 2018-10-20:
I'm not sure if you will be able to use that library with your board. It makes use of hardware SPI to talk to the NeoPixel which requires the use of pins P0_07, P0_08, P0_09 or pins P0_15, P0_17, P0_18. Those pins may not be available on your board (or if they are they may need to be shared with existing hardware). Unfortunately there does not seem to be a schematic available for your board (and the pins file is not complete). We would also need to know what other hardware you have attached to the board (in particular doyou have an LCD display and if so which one).

andyman198, 2018-10-20:
Let me get in touch with the vendor and see if they have any thoughts / advice. I will report back.

p3p, 2018-10-20:
Porting the NeoPixel lib over to LPC176x is on the todo list, will probably be bit banged so any pin can be used as having access to the spi hardware is unlikely.

gloomyandy, 2018-10-20:
Unfortunately the answer is maybe and it depends. There are only two hardware SPI devices (and the code you found uses hardware SPI). This board is essentially a Smoothieboard clone, so it almost certainly uses one of the SPI devices to access the onboard SD card (and possibly the one on the LCD as well), normally smoothieware will use the other SPI device to access the LCD (and possibly the external SD card), but it may also be using it to access the stepper drivers. However the current Marlin pins file does not seem to use the "standard" pins to access the VIKI2 display so I suspect it is using software SPI for the display and probably for the SD card as well. Do you have the VIKI2 working? Can you also use the external SD card reader? How do you have it attached to the control board? What we really need is a good description of all of the pins on the control board and which pins they go to on the LPC1769

andyman198, 2018-10-22:
To answer the questions, yes it’s connected in the standard way with the two cables and it is using the standard pins file. 

The connector does appear to be made up of all digital pins so maybe you are right with the software SPI. 

This shows the digital pins 
http://panucattdevices.freshdesk.com/support/solutions/articles/1000245711-x5-gt-install-guide

But as you can see here, what’s actually presented to the display is a combination of digital pins and SPI. 

https://s3.amazonaws.com/cdn.freshdesk.com/data/helpdesk/attachments/production/1043587235/original/Viki2_Wiring%20v1.2.pdf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJ2JSYZ7O3I4JO6DA%2F20181022%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20181022T122924Z&X-Amz-Expires=300&X-Amz-Signature=67ee999d6af9af4c3d1868d8410eecd88b7af0881e06d25712445a835fddb86d&X-Amz-SignedHeaders=Host&response-content-type=application%2Fpdf

I believe the SD card uses hardware SPI. I’m not sure on the trinamic drivers. Will report back.

andyman198, 2018-10-22:
In the sample SPI motor controller config from the azteeg website (http://panucattdevices.freshdesk.com/helpdesk/attachments/1049660327)

```
> # bigfoot settings
> motor_driver_control.alpha.spi_channel       0                # SPI channel 1 is sdcard channel
> motor_driver_control.alpha.spi_cs_pin        0.10             # SPI CS pin
```
So I think that confirms that hardware SPI 0 is for the Trinamics and SPI 1 is for the SDcard making the display a software SPI.

andyman198, 2018-10-22:
>           Porting the NeoPixel lib over to LPC176x is on the todo list, will probably be bit banged so any pin can be used as having access to the spi hardware is unlikely. Will probably need to get some to test with at some point.

Not sure if it helps @p3p, but happy to send a neopixel if you don't have one for testing

p3p, 2018-10-22:
Thanks for the offer @andyman198, already ordered a cheap ones to play with when I get round to it, no guarantees about when that will be though, pretty much all other features and bugs will be a higher priority than pretty colours ;)

andyman198, 2018-10-22:
I actually thought that the neopixels were all that was left on the LPC1768 lol! is there a tracking list? maybe I can help?

thinkyhead, 2018-10-26:
We could certainly implement our own Neopixel lib for LPC1768 on top of software SPI with configurable pins. How hard could it be, right?

andyman198, 2018-10-26:
The original adafruit library doesn't seem to even use SPI, just a timer with interrupts. (https://github.com/adafruit/Adafruit_NeoPixel) 
which would be the better way to go for the LPC176X?

msticninja, 2018-12-05:
Is anyone still working on this? Re-arm has very few pins, I'd love to save a couple by using Neopixels instead of discrete rgb channels.

boelle, 2019-02-21:
@andyman198 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.
Before posting a bug report please test with `bugfix-2.0.x` to check if problem is gone

p3p, 2019-03-07:
NeoPixels now supported on LPC176x

@andyman198 I don't have an LPC1769 so can't confirm the timings @ 120MHz but should be compensated for.

boelle, 2019-03-07:
👍

closed 2019-03-15 by p3p