MarlinFirmware/Marlin #12165: bugfix-2.0.x and VIKI2 + reArm

cdedwards, 2018-10-20:
After compiling lastest bugfix-2.0.x for my rearm board on the delta printer, viki2 display quit displaying anything. I've tried various contrast values from 1-250. no display changes at all. When I reflashed with my known working firmware.bin, the display started working again. I looked for changes in the st7565 file but could not find any

p3p, 2018-10-20:
Some of the default pin assignments may have changed, I know others still have the VIKI2 screen working, can you supply your config and pins file.

cdedwards, 2018-10-20:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2498352/Marlin.zip)

Here are my configuration.h _adv_h and pins file

p3p, 2018-10-20:
I can't open that zip file I think it's corrupt.

cdedwards, 2018-10-20:
it's actually an rar file

cdedwards, 2018-10-20:
[pins_RAMPS_RE_ARM.zip](https://github.com/MarlinFirmware/Marlin/files/2498386/pins_RAMPS_RE_ARM.zip)
here's a zip file

p3p, 2018-10-20:
just had to change the extension

p3p, 2018-10-20:
Do you know what pins the viki2 is connected to?, on lines 314, 315 they are being configured as SCK_PIN, MOSI_PIN but at that point they have not been defined. The default value for those would be P0_15, P0_18 for the lcd port, could you try it with those values, if using those pins and you use a sdcard on the display you will also need to define LPC_SOFTWARE_SPI.

cdedwards, 2018-10-21:
trying to print is painful.. really painful. I finally got a message about the drive needing to be formatted.  This is after about 15 minutes of being connected.

p3p, 2018-10-21:
What did you comment out? if you don't wan't to use sdcards disable `SDSUPPORT` if you want to disable the usb mass storage device, disable `USB_SD_ONBOARD`, and enable `USB_SD_DISABLED` in the pins file.

cdedwards, 2018-10-21:
#ifdef LPC_SD_ONBOARD
  #ifdef USB_SD_ONBOARD
    // When sharing the SD card with a PC we want the menu options to
    // mount/unmount the card and refresh it. So we disable card detect.
    #define SHARED_SD_CARD
    #undef SD_DETECT_PIN // there is also no detect pin for the onboard card
  #endif
  //#define SCK_PIN            P0_07
  #define MISO_PIN           P0_08
  //#define MOSI_PIN           P0_09
  #define SS_PIN             P0_06   // Chip select for SD card used by Marlin
  #define ONBOARD_SD_CS      P0_06   // Chip select for "System" SD card
#endif

cdedwards, 2018-10-21:
I had LPC_SD_ONBOARD enabled. so SCK_PIN and MOSI_PIN were defined in the above block.
commented them out so that the warning disappeared.

cdedwards, 2018-10-21:
//#define USB_SD_DISABLED
#define USB_SD_ONBOARD        // Provide the onboard SD card to the host as a USB mass storage device

#define LPC_SD_LCD          // Marlin uses the SD drive attached to the LCD
//#define LPC_SD_ONBOARD        // Marlin uses the SD drive on the control board

#ifdef LPC_SD_LCD
  #define SCK_PIN            P0_15
  #define MISO_PIN           P0_17
  #define MOSI_PIN           P0_18
  #define SS_PIN             P1_23   // Chip select for SD card used by Marlin
  #define ONBOARD_SD_CS      P0_06   // Chip select for "System" SD card
#endif

this works. display works. and the sd card on the re-arm works. The display SD does not

gloomyandy, 2018-10-21:
Did you need to define LPC_SOFTWARE_SPI? Can you post your (working) settings again so others with the same setup can see them.

rafaljot, 2018-10-21:
Probably I have similar issue with DUE+RAMPS-FD.
LCD doesn't work at all. Pins are correct.

p3p, 2018-10-21:
@rafaljot The issue here was pin allocation and lpc176x specific feature selection and will not be connected to your problem, you should create an issue with all the requested information from the template.

cdedwards, 2018-10-21:
#define LPC_SD_LCD // Marlin uses the SD drive attached to the LCD
//#define LPC_SD_ONBOARD // Marlin uses the SD drive on the control board

these are the only two lines needing to be changed. uncomment one, and add a comment to the other.  The SCK_PIN and MOSI_PIN were red herrings here. With LPC_SD_LCD defined, the proper pin definitions are made. otherwise there are improper pin definitions for a re-arm with VIKI2 made later on in the pin file. The other issue is that the system slooooooowwws way the hell down if you don't do it this way. LCD menu's are sluggish and the print I tried would have taken weeks to do since the printing was so slow.

Both the internal re-arm SD slot and the VIKI2 SD slot are now active.

[working pins_RAMPS_RE_ARM_with_VIKI2.zip](https://github.com/MarlinFirmware/Marlin/files/2499308/pins_RAMPS_RE_ARM_with_VIKI2.zip)

boelle, 2019-02-21:
@cdedwards problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-15 by p3p