MarlinFirmware/Marlin #12166: [sBase] Endstop interrupts is busted

forkoz, 2018-10-20:
pinmapping.h was removed and so it fails trying to check if a pin is interrupt capable.
I can of course comment out the checks but new users might have an issue. Unless I'm the only user :)

I know its not priority, just thought you should know.

p3p, 2018-10-20:
Considering the rewiring required on all the known boards in order to get at interrupt capable pins for the endstops you may be the only one, doesn't make this any less of a bug though. The macro version of the LPC1768_PIN_INTERRUPT function got removed from the pinmapping file for some reason (I'd tell you why but I have no memory of doing it ;) ).

p3p, 2018-10-20:
You can fix it temporarily by putting `#define LPC1768_PIN_INTERRUPT_M(pin) (((pin >> 8) & 0b1) != 0)` at the top of the `endstop_interrupts.h` file

forkoz, 2018-10-20:
thats much better than commenting it all out, thanks.

p3p, 2018-10-20:
sorry that's not going to work, I changed how pin information is stored as a constant and the old macro uses the old method, `#define LPC1768_PIN_INTERRUPT_M(pin) ((pin >> 0x5 & 0x7) == 0 || (pin >> 0x5 & 0x7) == 2)` will work though as all of port 0 and 2 support interrupts.

closed 2018-10-20 by forkoz

thinkyhead, 2018-11-05:
@p3p — Is this all good? : b12b53c74c6e7298427263754c2dfcefed50d3a8

p3p, 2018-11-05:
The 2 pins missing guards are a good find, no idea when they were added, but the Macro is already restored as part of the framework.