MarlinFirmware/Marlin #12171: [bugfix-2.0.x] - "Long" pause after probing

robustini, 2018-10-21:
### Description

There's way to decrease the time that elapses since the probe touches the bed and when it rises?
I can't find any parameter about that.
When the probe is activated (led turns blue) and the probe has already been retracted some time passes before the effector gets up, with Repetier there is no such boring pause.
This pause during "multitouch" creates a problem when using shifts reduced to a minimum to speed up the procedure, because the probe will deploy again when the effector is still low, resulting probing failed.
It stands still for almost half a second, and with the BLTouch (clone) creates problems, I would like the effector to get up almost immediately.

Video comparation, "M48" with "Marlin 2.0" (compiled with the latest commit of today) and "MarlinKimbra V4.3.7ß", both 32 bit.

https://youtu.be/rQ9fKRPST7k

https://youtu.be/ogVrrMDoihg

ManuelMcLure, 2018-10-21:
Do you have `PROBING_HEATERS_OFF` and `WAIT_FOR_BED_HEATER` turned on by any chance? If so, it may be waiting for the bed temperature to stabilize.

robustini, 2018-10-22:
No, i never use this options.
This issue is inside the Marlin code, in MarlinKimbra firmware this "pause" has been rightly removed by @MagoKimbra (the lead developer) only for the delta printer, where it is rightly not necessary.

MagoKimbra, 2018-10-22:
The problem is into void do_probe_move
Each time it activates and deactivates the bltouch.
```
// Deploy BLTouch at the start of any probe
#if ENABLED(BLTOUCH)
  if (set_bltouch_deployed(true)) return true;
#endif

// Retract BLTouch immediately after a probe if it was triggered
#if ENABLED(BLTOUCH)
  if (probe_triggered && set_bltouch_deployed(false)) return true;
#endif
```

And into void bltouch_command there is a delay safe_delay(BLTOUCH_DELAY);

This is useful to prevent the bltouch from being extracted before the printer can get back up, it's fine for printers with Z slow, but in the delta it's not needed.

MikeLud, 2018-10-22:
@robustini 

I might be how the HOMING_FEEDRATE_Z is set in the Configuration.h file

```
// Homing speeds (mm/m)
#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)

```

boelle, 2019-02-21:
@robustini problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead