MarlinFirmware/Marlin #12174: The Ender 3 configuration is set to 115200 baud rate

DominikPalo, 2018-10-21:
### Description

Is there a reason why the baud rate for Ender 3 (in the [example configuration](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/example_configurations/Creality/Ender-3/Configuration.h#L119)) is set to 115200 instead of "standard" 250000? As I know, the Ender 3 board contains a 16 MHz microcontroller and the most optimal baud rate for 16 MHz is 250000. I print using Octoprint and sometimes complex movements causes short lags, so I would like to increase the baud rate from 115200 to 250000 (it runs on Raspberry 3, so the CPU is not a problem).

Ludy87, 2018-10-21:
then try it and be happy 😉

DominikPalo, 2018-10-21:
Of course, I will try - but I'm just curious why is set as is :) If you look at the [main default config](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L126), there is 250000. I have a second printer with a Mini-Rambo board and it works on 250000 without problems.

TheSFReader, 2018-10-21:
I think the default Ender 3 configuration is aligned to the Ender3 official confg, which has 115200 so...

Ludy87, 2018-10-21:
Show you this https://www.reddit.com/r/3Dprinting/comments/8llebl/can_a_cr10ender_3_board_handle_250k_baudrate/

swilkens, 2018-10-21:
It was me that replied there on reddit, I've been using 250000 without issues ever since. 

There are some various mentions [1] on-line of sanguino boards not being happy on higher bitrates, but I haven't had a problem as of yet.

[1] https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_configuration/configuration.md

> The serial communication speed of the printer should be as fast as it can manage without generating errors. In most cases 115200 gives a good balance between speed and stability. Start with 250000 and only go lower if "line number" and "checksum" errors start to appear. Note that some boards (e.g., a temperamental Sanguinololu clone based on the ATMEGA1284P) may not be able to handle a baudrate over 57600. Allowed values: 2400, 9600, 19200, 38400, 57600, 115200, 250000.

swilkens, 2018-10-21:
I wonder though, if those short lags are actually caused by the buffer running out rather than the baudrate.

Ludy87, 2018-10-21:
this is the balance between quartz and chip.

swilkens, 2018-10-22:
In closing, I dug through to commits to find https://github.com/MarlinFirmware/Marlin/pull/10588

Here, the example config for this printer is first introduced to the codebase. It has 11520 baudrate. The commit mentions this was a replication of the stock setting [1] of the printer. We can also see [2] that this baudrate was the default setting from the manufacturer and I think it's logical to keep that setting so users switching over aren't faced with a sudden change in baudrate.

If this answers your question, please close the issue :)

[1] https://pastebin.com/vxV8etsJ
[2] https://github.com/Creality3DPrinting/Ender-3/blob/master/Ender-3%20Firmware%20(Marlin)/Ender-3%20(includes%20power%20failure%20resume-%20English)/Marlin/Configuration.h#L28

thinkyhead, 2018-10-26:
115.2K is safer and more reliable. Go ahead and use 250K if it works for you.

closed 2018-10-26 by thinkyhead

swilkens, 2018-10-26:
> 115.2K is safer and more reliable. Go ahead and use 250K if it works for you.

Out of curiosity, why is that - as the datasheets seem to imply differently? Is this because of software / driver support (i.e. 250k isn't a "traditional" baudrate in the sence of multiplication of 75 baud ttype)?

Just curious here, and I couldn't find a good clarification elsewhere.

thinkyhead, 2018-10-26:
Lower baud rates are less susceptible to noise and bad USB cables.

zakdma, 2019-06-13:
> 115.2K is safer and more reliable. Go ahead and use 250K if it works for you.

Official Ender-3 Pro use 250000 baud rate so it is better to change this settings in default Ender-3 configuration to 250000 for better compatibility.

TheSFReader, 2019-06-13:
I think the motherboard is different between the ender3 and ender3 pro, which can explain the difference between the baud rates.

InsanityAutomation, 2019-06-13:
Only issue i see with 250000 baud is cura occasionally has trouble connecting where it seems fine at 115200 on anything. These long standing bugs (and the lightspeed feedrate from the other issue) are a big part of why i havnt used it in a long time.... Otherwise yes, 250000 will likely produce less jitter and if my memory is correct thats why it was set as the typical default in the first place.

tablatronix, 2020-04-28:
Just to clarify, the creality controllers use ch340g usb uarts not FTDI.. so most of swilkens post does not apply perse.