MarlinFirmware/Marlin #12175: if TMC2130 is enabled it crashes very soon even before Init ()

mr-miky, 2018-10-21:
It's been more than a week since I've been trying to figure out why MARLIN does not want to work when change the config file from:
```
#define E0_DRIVER_TYPE A4988
// # define E0_DRIVER_TYPE TMC2130
```
to :
```
// # define E0_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2130
```
With the A4988 driver everything works fine the logo appears on the display and then from the main menu you can operate as usual.
With  TMC2130 driver does not even arrive at the Init ().
A species of exception is created that loop the STM32F4.
I have identified where it happens but it is not clear to me.

On **ultralcd_impl_DOGM.h**
```
// The current graphical page being rendered
u8g_page_t & page = ((u8g_pb_t *) ((u8g.getU8g ()) -> dev-> dev_mem)) -> p;
```
in particular, dev_mem is 0 when TMC2130 is enabled.
I bypassed the obstacle by declaring:

`u8g_page_t * page = 0;`

and then I set the value in:

**ultralcd_impl_DOGM.h**
```
// Initialize or re-initialize the LCD
// static void lcd_implementation_init () {
void lcd_implementation_init () {

if (page == 0) {
   page = & ((u8g_pb_t *) ((u8g.getU8g ()) -> dev-> dev_mem)) -> p;
}
```
I obviously modified the rest of the code to use the pointer.

With the A4988 driver everything works, with the TMC no but at least I can get to the setup ().

Attached to the debugger snapshots.
page is zero because the debugger is stopped at main () and lcd_implementation_init () has not yet been called
[image]

mr-miky, 2018-10-21:
Obviously no optimization (Debug / -Og)

thinkyhead, 2018-10-27:
Done.