MarlinFirmware/Marlin #12176: [Bugfix2.0] Requested Feature - Option for Raps128

bwgeez02, 2018-10-21:
There is no option for the Raps128 stepper in Configuration.h

 * Stepper Drivers
 *
 * These settings allow Marlin to tune stepper driver timing and enable advanced options for
 * stepper drivers that support them. You may also override timing options in Configuration_adv.h.
 *
 * A4988 is assumed for unspecified drivers.
 *
 * Options: A4988, A5984, DRV8825, LV8729, L6470, TB6560, TB6600, TMC2100,
 *          TMC2130, TMC2130_STANDALONE, TMC2208, TMC2208_STANDALONE,
 *          TMC26X,  TMC26X_STANDALONE,  TMC2660, TMC2660_STANDALONE,
 *          TMC5130, TMC5130_STANDALONE
 * :['A4988', 'A5984', 'DRV8825', 'LV8729', 'L6470', 'TB6560', 'TB6600', 'TMC2100', 'TMC2130', 'TMC2130_STANDALONE', 'TMC2208', 'TMC2208_STANDALONE', 'TMC26X', 'TMC26X_STANDALONE', 'TMC2660', 'TMC2660_STANDALONE', 'TMC5130', 'TMC5130_STANDALONE']
 */
//#define X_DRIVER_TYPE  A4988
//#define Y_DRIVER_TYPE  A4988
//#define Z_DRIVER_TYPE  A4988
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define E0_DRIVER_TYPE A4988
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988

leseaw, 2018-10-21:
LV8729V is raps128 driver.

GMagician, 2018-10-22:
@leseaw are you sure about your statement?

I read in more than one site that they use THB6128

https://www.reprap.me/media/manuals/RADDS_UserGuide_V2_reprapme.pdf
http://doku.radds.org/en/?s=raps128

leseaw, 2018-10-22:
sure :)
i sell this driver.
Only one problem:
old driver with green heat sink are with thb6129
New one with blue heat sink LV8729V.
So we need then two descriptions.

bwgeez02, 2018-10-22:
Mine are definitely THB6128 chips.

vmorrisPolar, 2019-02-14:
I'm not getting any movement yet, can anyone confirm what setting worked for them?

bwgeez02, 2019-02-16:
I left them all commented out.

boelle, 2019-06-22:
missing label

rimhanc, 2020-07-04:
> Mine are definitely THB6128 chips.

Hei, 

Can you please help me. I can't find this one too in marlin. I'm using RAPS 128. 

Kind Regards;

RH

thinkyhead, 2021-07-30:
If anyone can get support for this stepper driver added to Marlin, I'll be happy to merge it!