MarlinFirmware/Marlin #12177: Heated bed issue temperature steal increase during print

phugo40, 2018-10-21:
Hello guys,

I have an AM8 with ramps 1.4 and my heated bed is Anycubic ultrabase. I use mosfet too.

The weirdest thing is when I making PID for PETG (75º) the temperature of the heated bed stays stable, does not oscillate. For PLA (60º) the temperature rises to 90º / 100º

This is my second heated bed, brand new Anycubic ultrabase, my other bed (mk3) i had the same issue.

I've done everything, I think in the last case, it's some configuration of marlin.

Thank you

ejtagle, 2018-10-21:
I'd say this could be either a hw problem or a PID calibration issue

Ludy87, 2018-10-21:
Post your PID values for PLA and PETG and your config files

phugo40, 2018-10-21:
> Post your PID values for PLA and PETG and your config files

PID values:
PETG 75º
P308.97 I60.83 D392.32

PLA 60º
P226.26 I44.55 D287.29

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2499452/config.zip)

Ludy87, 2018-10-21:
While not the problem, why are you activating more drivers than you have?
Why do you use the Anet-Config?

    #endif PIDTEMPBED

change in

    #endif // PIDTEMPBED

phugo40, 2018-10-21:
> While not the problem, why are you activating more drivers than you have?
> Why do you use the Anet-Config?
> 
> ```
> #endif PIDTEMPBED
> ```
> change in
> 
> ```
> #endif // PIDTEMPBED
> ```
> Have you tried https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x or https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x

I bought anet an8 and set up for anet, then I was doing hw upgrades and parts and did not get pure marlin (for laziness). I'll change what you said. Thank you
 I did not try, what do you recommend?

Ludy87, 2018-10-21:
"Original" Anet A8 has `#define TEMP_SENSOR_BED 5`, disabled ` #define ENDSTOP_INTERRUPTS_FEATURE`, and disables the other drivers that are not connected.

With Ramps https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

closed 2018-10-26 by thinkyhead