MarlinFirmware/Marlin #12178: Bugfix2.0 and no numerical display for Y axis on LCD

RudyBenner, 2018-10-21:
Latest bugfix 2.0 does not display the Y axis position on the MKS MINI 12864 LCD. Just a blank.
Marlin 1.1.9 does show the Y axis correctly.
I have not tested bugfix 1.1.x

Y axis is displayed on Repetier Host and Printerface correctly.

[Bugfix 2.0.x configs.zip](https://github.com/MarlinFirmware/Marlin/files/2499368/Bugfix.2.0.x.configs.zip)
[Marlin 1.1.9 configs.zip](https://github.com/MarlinFirmware/Marlin/files/2499369/Marlin.1.1.9.configs.zip)

Lyr3x, 2018-10-22:
I can confirm. Exact same issue for me using the MKS MINI 12864 using bugfix-2.0: aa9202260dcf1add950afdb05942f3c0e89d9af4

Lyr3x, 2018-10-22:
I can confirm on build: aa9202260dcf1add950afdb05942f3c0e89d9af4

RudyBenner, 2018-10-25:
Not a big deal, since I operate from Repetier Host, but it should eventually be addressed.

thinkyhead, 2018-10-25:
Pretty odd! The code to draw it is the same for all graphical LCDs, and matches the technique used for X and Z. No special conditions for this display.

swilkens, 2018-10-25:
I did a bisect and found that the offending commit is: https://github.com/MarlinFirmware/Marlin/commit/67a3c5a16ad0c202352ef19b5993ac0c7f52862d

RudyBenner, 2018-10-25:
Please excuse my newbieness, how do I fix it?

RudyBenner, 2018-10-25:
In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_AVR\HAL_spi_AVR.cpp:40:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:460:6: error: #error "Alternative LCD_INFO_SCREEN_STYLE requires 20x4 Character LCD."
#error "Alternative LCD_INFO_SCREEN_STYLE requires 20x4 Character LCD."

Does not compile.

thinkyhead, 2018-10-26:
I’ll check it out. @RudyBenner could I borrow your configuration files for testing?

RudyBenner, 2018-10-26:
I posted both sets.

thinkyhead, 2018-10-26:
Oh yeah, there they are. I'll have a patch shortly, I'm sure…

RudyBenner, 2018-10-26:
20 lashes

RudyBenner, 2018-10-26:
[image]
[image]

thinkyhead, 2018-10-26:
Now patched! Counterintuitively, by not providing enough buffer space for the Z string, the Y value ended up with a `nul` as its first character.

closed 2018-10-26 by thinkyhead