MarlinFirmware/Marlin #12180: [bugfix-2.0.x] LIN_ADV is ignored

ikarisan, 2018-10-21:
Hi!

After facing some outer wall issues I thought that I might have chosen the wrong K-factor. So I printed the test pattern for values between 0.0 and 0.58. But every line looks the same! I guess that the set value is never used. 

[image]
[image]
[image]
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2499446/Configuration.zip)
[KFactor_TestPattern.zip](https://github.com/MarlinFirmware/Marlin/files/2499448/KFactor_TestPattern.zip)

I am using this version: 71e19baf691a5252841cca5871a29d7a65b62e5f

bwgeez02, 2018-10-22:
Try setting the start kfactor at 0 and end at 2.0 in 0.2 steps.

rb1205, 2018-10-22:
For bowden setups you'll probably end up higher than **2**, and your highest facotr in your test file is 0.6. No wonder you don't see any effect. go [here ](http://marlinfw.org/tools/lin_advance/k-factor.html)and build a custom test for your setup that goes at least to k=3

ikarisan, 2018-10-22:
Maybe I have to close this issue.....

A few weeks ago I printed the test pattern and the best value was around 0.30 and 0.35 (on direct extruder I was using 0.12). But now, suddenly, the optimal value seems to be around 0.85! I don't know why.... but the new value does not fix the "bad curve quality" :(

Lyr3x, 2018-10-23:
@rb1205 Thats absolutely not the case for my Bowden setup and Lin Adv. 1.5. Nearly for all types of Filament my k factor lays about 0.8

thinkyhead, 2018-10-26:
> But now, suddenly, the optimal value seems to be around 0.85! I don't know why

Maybe the Bowden tube is a little loose in the hotend. Try doing a "cold pull" while pulling out the Bowden tube from the hotend. Clear off all the material and jam the tube as far into the hotend as you can. See if it makes a difference. Check both ends.

boelle, 2019-02-21:
@ikarisan problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead