MarlinFirmware/Marlin #12183: Strange G26 behaviour with bugfix-2.0.x

cdedwards, 2018-10-21:
I enabled UBL on my Cube printer and started the Step By Step UBL from the menu. G29 P1, G29 to fill in the mesh, and then G29 S1 to save it.. When I attempt to Validate the mesh, I run into an issue. G26 heat's the bed/extruder and then attempts to prime the nozzle. I find that my Y motors are being driven backwards slowly instead of the nozzle pumping out any filament.. At this point my display is also frozen and I can't press a button either. 

I changed the heater temp's to 195 and 65 to match what I normally use. Preheat the bed and hotend up to temperature. and enter G26 C B70 H195 P1  to start the mesh validation. instant reboot of the mega2560.

boelle, 2018-10-22:
maybe attaching your config would enable the dev people to figure if its a config problem or not

put configuration.h and configuration_adv.h in a zip file and attach

[image]

cdedwards, 2018-10-22:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2501252/Configuration_adv.zip)

thinkyhead, 2018-10-26:
A reboot almost always indicates some feedrate is too high. See if `M502`, `M500` clears up the issue. And if not, check your max feedrate settings. This kind of problem is more prevalent when using micro-stepping of 32x and higher on AVR boards.

cdedwards, 2018-10-28:
I just enabled UBL on a friends Tevo Tornado for him. Attempted to run G26 again on his printer and found that it was printing very strange as the Z would go up and down instead of Y from the look of it. As for my printer, I reduced the number of points in the mesh which got me to the same point almost. It now starts the G26, but my Z is moving up and down and eventually it reboots after a couple of circles.

Roxy-3D, 2018-10-28:
My guess is some regression has been introduced into G26.   It might be a week or two...  But I'm starting fresh with a Formbot T-Rex3 printer.   I'm going to be bringing up UBL on it.   I'll be watching for this problem to happen as I bring up the printer.

Roxy-3D, 2018-10-28:
> A reboot almost always indicates some feedrate is too high.

Agreed.  But...   if you have gone crazy and specified a 15 x 15 mesh without having enough memory to handle it....  You will see reboots too.      

I'm almost thinking it makes sense to pull the mesh size back to 10 x 10.   I have two machines with 400mm beds, and 10 x 10 is more than enough to map the contour of the bed.

thinkyhead, 2018-11-07:
> if you have gone crazy and specified a 15 x 15 mesh without having enough memory to handle it

Marlin pre-allocates these buffers. A large pre-allocated (e.g., 900-1024 byte) array should be no problem in itself, unless as a result the compiler is warning that there's very little (e.g., less than 800 bytes) of SRAM remaining.

boelle, 2019-02-21:
@cdedwards problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead