MarlinFirmware/Marlin #12184: monitoring endstops during print prevents printing

boelle, 2018-10-22:
So i got updated to bugfix 2 yesterday

there must be a new setting somewhere i missed that monitors the endstops during print

since i use the pinda probe on my mendel90 with mk42 bed this one is triggered until 1mm up

so what is the setting called that monitors endstop after print is started? i need to disable it to print

boelle, 2018-10-22:
hmmm

since i use the pinda probe on my mendel90 with the mk42 bed there is no way arround this, it will remain triggered until about 1mm up (0,79+)

i have this in configuration_adv.h:

// If you want endstops to stay on (by default) even when not homing
// enable this option. Override at any time with M120, M121.
//#define ENDSTOPS_ALWAYS_ON_DEFAULT
sine the option is not enabled it should not have the endstops on when homing is done at start of print

a more interesting one:

  /**
   * This option allows you to abort SD printing when any endstop is triggered.
   * This feature must be enabled with "M540 S1" or from the LCD menu.
   * To have any effect, endstops must be enabled during SD printing.
   */
  //#define ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED
so since this feature is not enabled and the other is not i would expect to be clear

but my print is aborted due to a z endstop hit, since my Z endstops are top/max mounted and work fine i would rule them out, the only other endstop is the probe itself

my config is still the same as 3 days ago #12048 (comment)

boelle, 2018-10-22:
just started an 5 hour test print to dial in my retraction.... got some of those very fine hair stringing

did not end well: https://photos.app.goo.gl/XJtCcfhjdEumHWBz5

so think its back to my copy from 29/7 until we figure this one

boelle, 2018-10-25:
@thinkyhead 

do you think this could be a bug or maybe a config screwup on my part?

thinkyhead, 2018-10-26:
> since the option is not enabled it should not have the endstops on when homing is done at start of print

Not having endstops on when homing would be ridiculous.

> so since this feature is not enabled and the other is not i would expect to be clear

Indeed. Without `ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED` the code to abort an SD print on an endstop hit simply is not compiled in.

boelle, 2018-10-26:
that is the part i did not understand, ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED is not enabled but i still got aborted prints, 

of course we need endstops to be on when homing, but it was ENDSTOPS_ALWAYS_ON_DEFAULT i had a look at. the explanation says "If you want endstops to stay on (by default) even when not homing
enable this option."

the key words here are "when not"

so of course i do not enable this as i dont want them on when NOT homing

thinkyhead, 2018-11-04:
🤷‍♂️

boelle, 2018-11-04:
well i guess the only thing then is to download a copy once in a while and test to see if it still do a false hit z endstop during a print even thou the feture is turned off

and its the probe that does it since its triggered during the first few layers

and the serial out did say z endstop and since the z max endstops are not triggered it can only be the probe

thinkyhead, 2018-11-05:
See if the latest `bugfix-2.0.x` is better. An error was introduced in #9582 which I attempted to patch last night.

boelle, 2018-11-05:
will do later today, got a meeting in 30 mins and also cut out some cork vibration dampers

boelle, 2018-11-05:
short Q about config, is this correct for dual Z stepper driver?

```
#define X_DRIVER_TYPE  DRV8825
#define Y_DRIVER_TYPE  DRV8825
#define Z_DRIVER_TYPE  DRV8825
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE DRV8825
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE DRV8825
```

boelle, 2018-11-05:
did an test, this time print was aborted due to filament runout

but there is filament and M119 says the same

```
Recv: ok
[...]
Send: M119
Recv: Reporting endstop status
Recv: x_max: TRIGGERED
Recv: y_max: open
Recv: z_min: open
Recv: z_max: TRIGGERED
Recv: z2_max: TRIGGERED
Recv: filament: TRIGGERED
Recv: ok
[...]
```

InsanityAutomation, 2018-11-06:
I saw similar behavior not long ago but didn't have time to dig into it.

boelle, 2018-11-06:
well i'm disabling FILAMENT_RUNOUT_SENSOR and also ADVANCED_PAUSE_FEATURE along with NOZZLE_PARK_FEATURE, just to see if i can do a short test print

InsanityAutomation, 2018-11-06:
Just disabling filament runout did the trick here.

boelle, 2018-11-06:
oki, i thought that ADVANCED_PAUSE_FEATURE required both NOZZLE_PARK_FEATURE and FILAMENT_RUNOUT_SENSOR

but will try just FILAMENT_RUNOUT_SENSOR

boelle, 2018-11-06:
that did the trick also, if i had only made sure nozzle was primed :-D

boelle, 2018-11-06:
not sure if this matters but sometimes i get this compile error:

if i hit compile again it does not give the error

```
sketch\src\module\motion.cpp: In function 'report_current_position':

sketch\src\module\motion.cpp:180:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Users\boher\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/users/boher/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

InsanityAutomation, 2018-11-06:
That's an Arduino ide issue. I've seen it there, but not on platform io with vscode.

boelle, 2018-11-06:
can platform io output an hex file? i have been running it at some point and yes it has far better formatting of the config compared to arduino ide

but my printer is not directly attached so i need the hex file

InsanityAutomation, 2018-11-06:
Yeah, it has a build option that exports one.

boelle, 2018-11-08:
@thinkyhead i can confirm that the Z endstop(PROBE) hit issue has gone away but as @InsanityAutomation  said i now have to disable FILAMENT_RUNOUT_SENSOR or else that will stop the print

some extra Q's as i go through my config :

1)
using AUTO_BED_LEVELING_BILINEAR does it make sense to enable RESTORE_LEVELING_AFTER_G28 and G26_MESH_VALIDATION?

2) does PRINTJOB_TIMER_AUTOSTART make much sense when PRINTCOUNTER is disabled?

boelle, 2018-12-06:
@thinkyhead any news on this?

closed 2018-12-26 by boelle