MarlinFirmware/Marlin #12185: RepRap SmartController LCD not working with DUE + RAMPS-FD v2, bug fix-2.0.x

rafaljot, 2018-10-22:
LCD looks like not initialized. Serial communication with printer works.
Hardware is ok, Repetier works without any issue.

How to reproduce:
1.  download  bug fix-2.0.x
2. modify platform.ini file: 
env_default = DUE_USB
3. configuration.h:
#define MOTHERBOARD BOARD_RAMPS_FD_V2
...
#define REPRAP_DISCOUNT_SMART_CONTROLLER
4. Compile, upload

cjsoong, 2018-10-23:
@ @rafaljot I have no problem... check your settings.
[image]

rafaljot, 2018-10-23:
Hi, Could you send me or attach your configs ? Is it last bug fix-2.0 ?

tig33r, 2018-10-23:
Rotate connectors by 180 deg.

rafaljot, 2018-10-23:
@tig33r . Connectors are OK. It works with Repetier.

cjsoong, 2018-10-23:
@rafaljot 
[Marlin-bugfix-2.0.x_20181013.zip](https://github.com/MarlinFirmware/Marlin/files/2506155/Marlin-bugfix-2.0.x_20181013.zip)

Miran81, 2019-03-18:
@cjsoong do you use ramps-fd with arduino due ? ı download your files but your platformio.ini env=megaatmega2560 ?