MarlinFirmware/Marlin #12190: (1.1.9) Change filament (M600) - The extruder motor is locked for manual change of the filament

Mysel07, 2018-10-23:
I just installed Marlin 1.1.9. I have direct drive belted extruder with crown gear for manual insertion filament. When I run on the LCD display, the function "Change filament" (or M600 from Pronterface), my 3D printer does everything right, but Extruder motor stay locked and and I can not change filament. In Marlin 1.1.7 i have no problem and configuration is basically the same.

In configuration.h is #define NOZZLE_PARK_FEATURE enabled.

configuration_adv.h is here : 
#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE         15  // (mm/s) Initial retract feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH            3  // (mm) Initial retract.
                                                  // This short retract is done immediately, before parking the nozzle.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE      15  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_ACCEL         20  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH        0  // (mm) The length of filament for a complete unload.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
                                                  //   Set to 0 for manual unloading.
  #define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE   10  // (mm/s) Slow move when starting load.
  #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH     0  // (mm) Slow length, to allow time to insert material.
                                                  // 0 to disable start loading and skip to fast load only
  #define FILAMENT_CHANGE_FAST_LOAD_FEEDRATE   15  // (mm/s) Load filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_FAST_LOAD_ACCEL      20  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_FAST_LOAD_LENGTH     0  // (mm) Load length of filament, from extruder gear to nozzle.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
  #define ADVANCED_PAUSE_CONTINUOUS_PURGE         // Purge continuously up to the purge length until interrupted.
  #define ADVANCED_PAUSE_PURGE_FEEDRATE        15  // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_PURGE_LENGTH          0  // (mm) Length to extrude after loading.
                                                  //   Set to 0 for manual extrusion ?
                                                  //   Filament can be extruded repeatedly from the Filament Change menu
                                                  //   until extrusion is consistent, and to purge old filament.

                                                  // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_RETRACT_LENGTH       0  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_DELAY                0  // 5000(ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         0  // (mm) An unretract is done, then this length is purged.

  #define PAUSE_PARK_NOZZLE_TIMEOUT           120 // (seconds) Time limit before the nozzle is turned off for safety.
  #define FILAMENT_CHANGE_ALERT_BEEPS           5 // Number of alert beeps to play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change.
  #define PARK_HEAD_ON_PAUSE                     // Park the nozzle during pause and filament change.
  #define HOME_BEFORE_FILAMENT_CHANGE           // Ensure homing has been completed prior to parking for filament change

  //#define FILAMENT_LOAD_UNLOAD_GCODES           // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
  //#define FILAMENT_UNLOAD_ALL_EXTRUDERS         // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif

Of course, I have turned on a feature #define PAUSE_PARK_NO_STEPPER_TIMEOUT , but this feature should only be stay turned on motors Y,Y and Z. No E- extruder motor.

Please how can I turn off the E motor for manual replace the filament.
Thank You

tcm0116, 2018-10-25:
Can you please attach your entire configuration files in a zip file?

Mysel07, 2018-10-26:
Thank You for your reply. Config files is here.
[Conf.files.zip](https://github.com/MarlinFirmware/Marlin/files/2517823/Conf.files.zip)

knexfour, 2019-01-01:
I am having the exact same issue and I'm really banging my head on it. Were you able to discover anything? Keeping the Extruder stepper locked makes changing filament pretty much impossible. Thank you

fatmann66, 2019-01-20:
in marlin_main.cpp about line 6838:
 // Disable extruders steppers for manual filament changing (only on boards that have separate ENABLE_PINS)
    #if E0_ENABLE_PIN != X_ENABLE_PIN && E1_ENABLE_PIN != Y_ENABLE_PIN
      disable_e_stepper(active_extruder);

so if you cannot disable just your extruder then it leaves all steppers enable

to test:

issue command: M17 
this enables steppers.
then issue command: M18 E
if your extruder and all steppers are still enabled then try : M18 Y
in my testing  M18 Y disables all steppers  not just Y. this indicates that they board is not cable of disabling just E and is why you are having the issue.

the above is there for a reason and bypassing it may cause issues and / or damage to your printer
(  THIS IS NOT SMART!! DON"T DO IT!! )  YOU HAVE BEEN WARNED!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
in my case I only change filament in between prints so losing  place of print location is not an issue
workaround:  

at approximate line number 6830 you find line     
"do_pause_e_move(unload_length, FILAMENT_CHANGE_UNLOAD_FEEDRATE);"
on the next line add the following code on it own line:
disable_all_steppers(); 

I normally tag any changes  with  initials  so a //FM66 at the end  so I can find it again.

this should disable ALL steppers once the unload is complete.

I don't do a " change filament" as I have  had the mentioned stepper lock issue  and prefer to just do an unload then a load when I'm ready. so you may have to add the code to another location for  the change to work

Regards,
Fatmann66

Mysel07, 2019-01-21:
Thank you, I will test it.

boelle, 2019-02-21:
@Mysel07 problem solved in latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead

VanessaE, 2019-10-22:
Can you point to where this was fixed?  It's still a problem at e604f767

fatmann66, 2019-10-22:
It wasn't fixed. His comment was it was addressed in marlin 2.0 not 1.19

VanessaE, 2019-10-22:
Hence my question, as I'm on bugfix-2.0.x.  It isn't fixed, at least not as of e604f76.