MarlinFirmware/Marlin #12191: Redefinable PINS for E1 (extruder 2) in "pins_RAMPS.h"

RepRapMe, 2018-10-23:
### Description

All firmwares for Creality CR-10 printers use the board RAMPS_14_EFB and in the "pins_RAMPS.h" the PIN settings for E1 (extruder 2) is already defined.
However the mainboard of the Creality CR-10 printers do not have the extra stepper driver for E1 soldered to the board, and if you want to have a dual extruder, you need to redefine the PINs for E1 to the AUX pins on the board.
To redefine the PINs you could change the file "pins_RAMPS.h", but it will be more convenient to define them in the "configuration.h".

### Steps to Reproduce

I'll suggest to add this to the "pins_RAMPS.h":

#ifndef E1_STEP_PIN
  #define E1_STEP_PIN      36
#endif
#ifndef E1_DIR_PIN
  #define E1_DIR_PIN       34
#endif
#ifndef E1_ENABLE_PIN
  #define E1_ENABLE_PIN    30
#endif

#### Additional Information

Ludy87, 2018-10-23:
The `configuration.h` is a "developer file" and the `pin_**.h` is a "fixed" configuration of the PINs that are set only once.

InsanityAutomation, 2018-10-23:
The CRX ships with a 2.1 board that does have these soldered in. Though the MKS 1.4 pin definition is more accurate for these than ramps, as pin 7 is used for the second hotend output on the creality boards as well.

Roxy-3D, 2018-10-23:
It would seem to me the best path forward would be to create a Creality CR-10 pins file.

tcm0116, 2018-10-25:
The new CR-10 pins file could include pins_RAMPS.h and the #undef and #define as needed. See pins_RAMPS_PLUS.h and pins_RAMPS_ENDER_4.h as examples.

InsanityAutomation, 2018-10-25:
Or just make it a duplicate name for including pins_MKS_GEN_13.h - might be clearer with it's own file, but don't want to create a ton of duplicate stuff either.

thinkyhead, 2018-10-26:
If CR-10 is shipping with something unique, then it should get its own pins file, set up exactly as Thomas describes.

InsanityAutomation, 2018-10-26:
Easy enough. I can whip that together pretty quick. I'll add in the servo pin definition for how the bltouch is added to enders with an itndef on the buzzer.

TheSFReader, 2018-10-26:
I think the Ender 3 uses the same board, doesn't it ? would this pins file need to be linked from the ender 3 conf ?

InsanityAutomation, 2018-10-26:
No, ender 3 is a melzi, same as the old cr10. Ender 4 is a different board yet, but similar to the cr10s ramps clone.

InsanityAutomation, 2018-10-26:
FYI #12229

closed 2018-11-02 by thinkyhead