MarlinFirmware/Marlin #12192: Feature Request - Query Axis Mode

FormerLurker, 2018-10-23:
### Description

I would like to request that a gcode be added or altered so that the current axis mode can be queried.  Ideally we could use the gcode to figure out if the XYZ or E axis is in relative or absolute mode.

The lack of this feature creates a profound problem for serial based software like Octoprint (and many others) where assumptions must be made about the current state of the printer.  These assumptions can and often do lead to serious issues like running the nozzle into the bed, and create serious programming challenges for developers.

I am agnostic about which gcode is used, but for example let's assume M145.  It need not take any parameters, I don't believe.  The response could be as simple as this, where E is the extruder and A is the XYZ axis:

```
E:RELATIVE A:ABSOLUTE
```
Thank you for reading this suggestion!

Roxy-3D, 2018-10-23:
It maybe valuable to also report how the various tools are configured also.   For example, and IDEX machine might have the 2nd extruder parked.  Or it might have it in a duplication mode.   It probably would be a good idea to let host software determine what the configuration is.

FormerLurker, 2018-10-23:
Very good points.  I can imagine other features that would be useful as well, such as retrieving the current units (MM vs some other units like inches, though I haven't heard many complaints that I assume MM by default in my software).  Ideally any settings configured via gcode could also be retrieved via gcode.  It just so happens that the axis modes are the most important from my perspective to prevent some common issues and programming problems.  

However, I can think of some other various data points that would be useful, such as the build volume in absolute coordinates (including any negative axis, for example ymin=-3 for MK2 and MK3), or the configured retract/detract/lift configuration set by M207 and or M208.  Given some time I could probably come up with more ideas regarding other info that would be useful.

If you need specific use cases I'd be more than happy to provide them.  For now I am assuming the value of this data is somewhat obvious.

thinkyhead, 2018-10-26:
Slicers ensure relative or absolute mode by sending the appropriate command, because the state of the machine and its defaults aren't known, and it should always be possible to send G-code by a dumb terminal or print from SD and not have to rely on hosts.

FormerLurker, 2018-10-26:
@thinkyhead, I've seen plenty of start gcode that doesn't include g90/G91.  That's why I requested the ability to query the current axis mode.  Also, I'm not sure what you mean by 'not having to rely on hosts', as I only wish to query the current mode just like I can request the current position via M114.

comps, 2018-10-26:
I believe the added value is mainly visible when pausing/resuming. A host software might want to pause on user request, then allow the user to jog the printhead around, unclog a nozzle, etc., and then continue streaming the original file.

This is essentially impossible without assuming the modal states the gcode file's beginning has set up (G90 vs G91, M82 vs M83, G0 vs G1, etc.).

For example - imagine a gcode file is running in G91, the host software pauses it, does its own G91 for easy jogging, and when resuming, does G90 because it assumed that the original gcode was in absolute coordinates, making the job fail.
If it did nothing, then an originally G90 job would be now running in G91.

It's also not a revolutionary idea.
https://github.com/gnea/grbl/blob/master/grbl/report.c#L274
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#status-reporting
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#10---status-report-mask
http://linuxcnc.org/docs/devel/html/gcode/m-code.html#mcode:m70-saved-state and M71/72/73

Gcode that doesn't specify its units (at the beginning) is not really something we can do much about as there is no right answer for what units it should use.

FormerLurker, 2018-10-30:
@comps, pretty much this.  I deal with issues weekly for my software (Octolapse) stemming from exactly this problem.  I can see no good way out of the issues I receive without adding some new firmware capabilities.

Regarding the units, this potential problem is WAY less severe from my perspective.  I've only ever received one issue regarding units since everyone seems to use MM.

I see the linked issue regarding reporting axis details via M114, but I think this will cause host software developers some substantial issues since they will need to change their REGEX expressions to work with any new additional info.  Using M114 would still be viable, but it would probably be simpler to use a new gcode.

OutsourcedGuru, 2018-11-07:
This is an excellent suggestion and much-needed.

foosel, 2018-11-12:
As with a lot of current internal firmware state that so far is basically write only (see also fan speeds, current tool, feed rate, feed and flow rate modifiers, all axes' E values, ...), being able to query this via serial would be helpful for hosts.

thinkyhead, 2018-11-12:
@foosel — Do you currently use the `M408 S0` command provided by RepRapFirmware and MK4Duo (primarily for the benefit of PanelDue)? https://reprap.org/wiki/G-code#M408:_Report_JSON-style_response That is something that I've started to work on in #10610.

FormerLurker, 2018-11-12:
@thinkyhead, if more information could be obtained via M408 (axis modes specifically), that would definitely work for me.  I prefer json anyway.  I'm guessing adding more fields wouldn't present a problem.

foosel, 2018-11-13:
@thinkyhead no because it's not generally usable (yet)

thinkyhead, 2018-11-13:
Yeah, I suppose people only compile `M408` into RepRapFirmware when they have a PanelDue, and then they probably wouldn't be using OctoPrint at the same time. I'm implementing it to be more generally usable, and I will make sure that there's an option to include it even when not using the `PANELDUE` option. It has a bunch of pre-defined keys, but of course we can always add more.

FormerLurker, 2018-11-14:
@thinkyhead, thanks for putting so much thought and effort into this!  I am really excited to start using M408.  It will have a huge positive effect on my code, and will greatly benefit user experience.

If you can, I'd love to know what keys you are planning to add.  I could provide a ranked wish list if that would be helpful.

Domush, 2022-02-07:
https://github.com/MarlinFirmware/Marlin/pull/23689 

COMPACT_STATUS_REPORTS
This should address the lack of reporting of modes and other critical information to hosts.

It changes the output of M154 and, optionally, M114.
`
/**
 * Compact Status Reports
 *
 * Use an abbreviated, advanced status report designed to relay maximum information to the host
 * in the smallest number of bytes (not very human-readable, designed for error-free host UI updates)
 *
 * Example: |#|X10.000:Y100.000:Z0.5|X20.000:Y120.000:Z10.5|1|0|1|2|110|2
 * Legend:
 *  Status header [|#], Work coords (G92..), Machine coords (G53), Metric [1] (G21), Absolute pos. mode [0] (G90),
 *  Tool #[1] (active_extruder), Coord set [2] (G55), Feed rate override [110]% (M220 S110), Status: MF_WAITING [2]
 *
 * As you can see, quite a bit of info is transmitted to hosts which support it, saving multiple verbose queries.
 */
#define COMPACT_STATUS_REPORTS
#if ENABLED(COMPACT_STATUS_REPORTS)

  // M114 will return compact reports instead of the usual verbose report
  // #define M114_USES_COMPACT_REPORTS // CAUTION - May break host UI's which look for M114's verbose output

  // #define GRBL_COMPATIBLE_STATES    // Report GRBL-equivalent states instead of Marlin states
#endif
`