MarlinFirmware/Marlin #12193: [2.0.x] 3a46212 Prevents movement on all axes when hotend is cold

swilkens, 2018-10-23:
### Description

https://github.com/MarlinFirmware/Marlin/commit/3a46212 is related to prevent extrusion while the hotend is cold, specifically it seems to be intended to clarify more clearly why extrusion is prevented.

However, on recent 2.0.x-bugfix movement on ALL axis is prevented. Every movement request through the LCD menu brings up a "Hotend too cold" message. This includes after homing.

I am not able to check if heating up the hotend releases all axes as my hotend is currently in pieces.

### Steps to Reproduce

1. Flash current bugfix (e229b831fab7b3e789631ea004640b7c82555253)
2. Attempt to move any axis through the LCD menu
3. Notice the "Hotend too cold" message
4. Home all axes
5. Repeat, notice the same message again

**Expected behavior:**
I would expect movement on all axes except the extruder

**Actual behavior:**
Movement on all axes is prevented

#### Additional Information
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2506873/configuration.zip)

swilkens, 2018-10-23:
Looks like here (https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/ultralcd.cpp#L3228) the submenu used for manual E movement is the same submenu as is used for X, Y and Z.

Thus, when the hotend is cold, the submenu for moving X, Y and Z is also modified with the warning message - thereby preventing the movement. A check if the user is moving E - not X, Y or Z - should catch this.

pietrobaricco, 2018-10-24:
same issue here

rmoravcik, 2018-10-24:
Something like this should fix it

```
diff --git a/Marlin/src/lcd/ultralcd.cpp b/Marlin/src/lcd/ultralcd.cpp
index 4f710f9ea..49bd161f5 100644
--- a/Marlin/src/lcd/ultralcd.cpp
+++ b/Marlin/src/lcd/ultralcd.cpp
@@ -3226,7 +3226,8 @@ void lcd_quick_feedback(const bool clear_buttons) {
       }
     }
     #if ENABLED(PREVENT_COLD_EXTRUSION)
-      if (thermalManager.tooColdToExtrude(eindex >= 0 ? eindex : active_extruder))
+      if ((axis == E_AXIS) &&
+          (thermalManager.tooColdToExtrude(eindex >= 0 ? eindex : active_extruder)))
         MENU_BACK(MSG_HOTEND_TOO_COLD);
       else
     #endif
```

swilkens, 2018-10-25:
@rmoravcik's commit fixes the issue on my (single extruder) printer, closing this

closed 2018-10-25 by swilkens