MarlinFirmware/Marlin #12204: G1 motion ignores maximum feed rate

PeterGrace, 2018-10-24:
### Description

G1 motion ignores maximum feed rate specified 

### Steps to Reproduce

1. Define maximum feed rate for any axis via M203 (e.g., M203 Z100)
2. Issue a G1 with feed rate higher than maximum (G1 Z100 F10000)
3. Observe as motor seizes as it attempts to comply

**Expected behavior:**

G1 motion will only move as fast as maximum feed rate specified in M203

**Actual behavior:**

G1 motion attempts to comply with F setting

#### Additional Information

I noticed this issue as I ran into a string of multiple hot ends jamming.  Upon review I found that even though my maximum extruder rate was set at 150mm/m in marlin, my slicer was trying to push filament considerably faster than that, causing extreme backpressure on the bowden assembly and eventually leading to tube deformation in the hot end assembly.  I'm assuming Marlin should not honor arbitrary feed rates that are in excess of the posted maximum, otherwise why does M203 exist?

Configuration files attached.

[i3pro-config.zip](https://github.com/MarlinFirmware/Marlin/files/2510857/i3pro-config.zip)

PeterGrace, 2018-10-24:
I'm sorry, I'm using 1.1.9-release.

swilkens, 2018-10-24:
Still seems fine to me, is it possible you think the motor is seizing (slow pulsed sound) because of the difference in units?

`M203 Fxxx` takes `F` in `mm/s` while `G0` and `G1` take `F` in `mm/m`

PeterGrace, 2018-10-24:
I was unaware that M203 and G1 used different timebases.  I will re-test with a more severely contrived example, perhaps M203 Z1 and then try G91 followed by G1 Z100 F10000 -- if I'm in error about this bug, the gantry should move at 1 mm/s max.

PeterGrace, 2018-10-24:
Yes, I am mistaken.  The unit differences are what caught me off guard.  Apologies.

closed 2018-10-24 by PeterGrace