MarlinFirmware/Marlin #12212: Feature Request - AC Phase Control for HeatBed

svs072477, 2018-10-25:
For smooth control of the temperature on AC 220v HeatBed , there is a circuit AC Phase Control.
https://playground.arduino.cc/Main/ACPhaseControl
Is it possible to add to the Marlin firmware management for the scheme, and the possibility of PID tuning for AC 220V HeatBed?

kAdonis, 2018-10-25:
Thats exactly what an AC-SSR is doing but it needs only one arduino pin 
I dont think this is needed

Roxy-3D, 2018-10-25:
> I dont think this is necessary

Agreed.   And I don't like the idea of connecting up logic to the 110 VAC or 220VAC line and bringing it to the microprocessor.   That is just asking for trouble.

The PID algorythms don't watch the wave form of the power being supplied.   But there is no need for that because the heatbeds change temperature so slowly.

svs072477, 2018-10-25:
The connection to the microcontroller is made via galvanic isolation, there are no problems with this. But if the SRR allows you to easily maintain a predetermined temperature, then perhaps this scheme is valid superfluous in this case

kAdonis, 2018-10-25:
I am using a cheap SSR for my heatbed, inside the SSR is an optocoupler and it is switching AC at zero-crossing
The bed is heated by a 230V AC/ 450W  silicone mat, working flawlessly with PID for more than a year now

svs072477, 2018-10-25:
ok, the question is closed

closed 2018-10-25 by svs072477

Roxy-3D, 2018-10-25:
I have a Fortek SSR controlling a 110 VAC Silicon Heat Bed on two of my printers.   You can get the Fortek SSR's on eBay for $5 or $6 including shipping.   BUT...  You need to go many times over the current you need because many of these devices are counterfiet and not really made by Fortek.    This one will work to drive your heatbed:   https://www.ebay.com/itm/1pcs-FOTEK-Soild-State-Relay-SSR-75DA-H-75A-DC-AC-90-480VAC-3-32VDC/283082945890?hash=item41e90ef562:g:3KAAAMXQWzNSjGyE:rk:29:pf:0

svs072477, 2018-10-25:
Thank you, apparently people who complained to me about Fortek SSR, just got fake

InsanityAutomation, 2018-10-25:
Same ones I use. Take ones have the same ic, just undersized to the label.... Hence the go oversized comment. If you want to add safety, get an open on rise snap disc thermal couple and put it in line to power and mounted under the bed.

Roxy-3D, 2018-10-25:
> Thank you, apparently people who complained to me about Fortek SSR, just got fake

Pretty much...  Any Fortek SSR you buy on eBay is counterfeit.   If I really knew it was a Fortek device I was buying...  I would be OK with a 10 or 15 amp device.   But because there is no way to know what it really is why buying on eBay, you are forced to go many times higher than what you need.

Don't say "I need 5 amps of current, so a 10 amp device will have plenty of margin."   That won't work when buying SSR's on eBay.

AnHardt, 2018-10-26:
PWM on a AC-SSR does work when the PWM frequency is (much) lower than the AC frequency.

With Marlins **normal** heater pwm at about 8Hz we get 0 to 2\*AC_FREQUENCY / 8 half waves. (100/8 = 13 --- 120/8 = 15 --- thats not great, but does work well)
With a **hardware** pwm it will not work well because the frequency is too high. When you get multiple 'triggers' during one half wave, you'll get a "permanent on" as soon the duty cycle is different from 0.

svs072477, 2018-10-26:
Thanks, now I understand

G-Pereira, 2019-08-30:
Well, yes if the pwm is much below 50Hz it works, but not ideally. It also shouldn't be a multiple of 50.
If the pwm signal is not in phase with mains power it will "trigger" full waves.
In a perfect scenario the PWM pulse would be 50Hz and in phase with mains voltage. I am also concerned with this. I already use a circuit to detect zero crossings in my DIY reflow oven so I can help with hw and sw but I've looked to marlin's source code and it's not readable in my opinion.