MarlinFirmware/Marlin #12213: Personal Config Ver numbering

tip32a, 2018-10-25:
I have converted my Folger-tech printer from Belts to screws in both x and y and added a modified version of a wades extruder.  I am trying to dial in the settings in Marlin Configure.h.  
I am using 1.1.8 FW
My Question
Is there a way to add a small comment (a date or rev number (maybe 6 to 8 digits)) that could pop up on the LCD on some screen that I could customize each time I update the firmware with my settings.

This way I can always know what modifications are loaded.  I would keep a journal that documents the changes and their results.

Thank you.

swilkens, 2018-10-25:
You might use `#define CUSTOM_MACHINE_NAME "foo"` for this ?

it's defined in `Configuration.h`

ManuelMcLure, 2018-10-25:
You could also override `STRING_SPLASH_LINE1` which by default gives the version of Marlin (i.e. "1.1.9" or "bugfix-1.1.x").

tip32a, 2018-10-25:
Thank you both Swilkens and ManuelMcLure.
I will test both of these this evening.

marcio-ao, 2018-10-29:
@tip32a: What we do is we edit "Marlin/src/inc/Version.h" and change:

```
#define SHORT_BUILD_VERSION "bugfix-2.0.x"
```

To:

```
#define SHORT_BUILD_VERSION "bugfix-2.0.x" LULZBOT_FW_VERSION
```
Then in "Configuration.h" we add the following:

```
#define LULZBOT_FW_VERSION ".1"
```

closed 2018-11-02 by thinkyhead