MarlinFirmware/Marlin #12216: [2.x] compile failure - Teensy ADC lib not compatible with AVR

fiveangle, 2018-10-26:
#12093 broke Teensy AVR platforms due to Teensy ADC library not compatible

Configs: https://github.com/fiveangle/Marlin/tree/bf2bisect1/Marlin

```
Compiling .pioenvs\at90usb1286_dfu\lib783\ADC\ADC_Module.cpp.o
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.cpp:30:0:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.h:41:2: error: #error "Sorry, this ADC library only works on ARM-based Teensy boards."
#error "Sorry, this ADC library only works on ARM-based Teensy boards."
^
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.h:49:0,
from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.cpp:30:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:48:2: error: #error "Board not supported!"
#error "Board not supported!"
^
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.h:49:0,
from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.cpp:30:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:336:2: error: #error "F_BUS must be 108, 60, 56, 54, 48, 40, 36, 24, 4 or 2 MHz"
#error "F_BUS must be 108, 60, 56, 54, 48, 40, 36, 24, 4 or 2 MHz"
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:499:23: error: 'ADC_REFERENCE' has not been declared
void setReference(ADC_REFERENCE ref_type);
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:764:20: error: 'ADC_INTERNAL_SOURCE' has not been declared
int analogRead(ADC_INTERNAL_SOURCE pin) __attribute__((always_inline)) {
^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\at90usb1286_dfu\lib783\ADC\ADC.cpp.o] Error 1
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.cpp:32:0:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:48:2: error: #error "Board not supported!"
#error "Board not supported!"
^
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.cpp:32:0:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:336:2: error: #error "F_BUS must be 108, 60, 56, 54, 48, 40, 36, 24, 4 or 2 MHz"
#error "F_BUS must be 108, 60, 56, 54, 48, 40, 36, 24, 4 or 2 MHz"
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:499:23: error: 'ADC_REFERENCE' has not been declared
void setReference(ADC_REFERENCE ref_type);
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:764:20: error: 'ADC_INTERNAL_SOURCE' has not been declared
int analogRead(ADC_INTERNAL_SOURCE pin) __attribute__((always_inline)) {
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h: In member function 'void ADC_Module::continuousMode()':
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:636:41: error: 'setBit' was not declared in this scope
setBit(ADC_SC3, ADC_SC3_ADCO_BIT);
^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\at90usb1286_dfu\lib783\ADC\ADC_Module.cpp.o] Error 1
 [ERROR] Took 374.01 seconds
```

p3p, 2018-10-26:
so platformio is being "helpful" comment out line 49 (`#include <adc.h>`) of `LPC1768/HAL.h`, it fixes the build, you will need to add whatever library it is pulling in to the ignore list for your platform because it spots that header in code it isn't going to build as it's the wrong platform and gets confused.

fiveangle, 2018-11-07:
#12311 implemented to work around PIO LDF issue

closed 2018-11-07 by fiveangle

fiveangle, 2018-11-19:
@p3p - Hey Chris, you may want to recheck to see if it's still worth manually specifying all libraries as a w/a for LDF because it seems there indeed was a bug in the src_filter function resolved in PIO 3.6.3b1