MarlinFirmware/Marlin #12217: bugfix_2.0 G28 homing Z_HOMING_HEIGHT not working at all

blairq, 2018-10-26:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-10-26:
So, we do have a sanity check for inverting and pullups to be set a certain way when using sensorless homing. Was that sanity check not correct? Or is it possible that the TMC’s threshold error pin is not connected to the proper endstop plug, or that sensorless homing sensitivity was set so low as to always be triggered?

blairq, 2018-11-06:
wiring was fine.

i think marlin is goeing away from using tmc on z axis so that may be the reason there are no checks for this.
there may  be a check like in x and y sensorles homing who tells you must set z_(max||min)_endstop_inverting must be on if using tmc sensorless homing (on Z)

thinkyhead, 2018-11-07:
```cpp
#if ENABLED(SENSORLESS_HOMING)
  . . .
  #elif Z_SENSORLESS && Z_HOME_DIR == -1 && (!Z_MIN_ENDSTOP_INVERTING || DISABLED(ENDSTOPPULLUP_ZMIN))
    #error "SENSORLESS_HOMING requires Z_MIN_ENDSTOP_INVERTING and ENDSTOPPULLUP_ZMIN when homing to Z_MIN."
  #elif Z_SENSORLESS && Z_HOME_DIR ==  1 && (!Z_MAX_ENDSTOP_INVERTING || DISABLED(ENDSTOPPULLUP_ZMAX))
    #error "SENSORLESS_HOMING requires Z_MAX_ENDSTOP_INVERTING and ENDSTOPPULLUP_ZMAX when homing to Z_MAX."
  . . .
#endif
```

boelle, 2019-02-21:
@blairq problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-11 by shitcreek