MarlinFirmware/Marlin #12226: Unable to compile Marlin 2.0 for STM32F4 board in Arduino IDE

bipsendk, 2018-10-26:
### Description

Unable to compile Marlin 2.0 for STM32F4 board

### Steps to Reproduce

Install STM32Generic in Arduino IDE 1.9.0-Beta (and also have Arduino Due and Zero added in board manager)

Copy example config file fra STM32F4 to marlin directry

Select "Black F407VG/VE/ZE/ZG Boards" from boards

Try to compile

**Expected behavior:**

Compile successful

**Actual behavior:** [What actually happens]

Fails with this command (verbose output)

"C:\\Users\\brian.ipsen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DF_CPU=168000000L -mthumb -DSTM32GENERIC -DRAM_LENGTH=131072 -DFLASH_LENGTH=524288 -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -DSTM32F4 -DARDUINO=10900 -DARDUINO_BLACK_F407VE -DARDUINO_ARCH_STM32 -DSTM32F407VE -DHSE_VALUE=8000000 -DMENU_USB_SERIAL -DMENU_SERIAL_AUTO=SerialUSB "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\cores\\arduino/stm32" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\cores\\arduino/usb" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/CMSIS" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/CMSIS_Inc" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/CMSIS_Src" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/HAL_Inc" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/HAL_Src" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/stm32_chip" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\cores\\arduino" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\variants\\BLACK_F407VE" "-IC:\\Users\\brian.ipsen\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\libraries\\SPI\\src" "C:\\Users\\BRIAN~1.IPS\\AppData\\Local\\Temp\\arduino_build_368782\\sketch\\src\\HAL\\HAL_STM32\\watchdog_STM32.cpp" -o nul
C:\Users\BRIAN~1.IPS\AppData\Local\Temp\arduino_build_368782\sketch\src\HAL\HAL_STM32\watchdog_STM32.cpp:30:25: fatal error: IWatchdog.h: No such file or directory

   #include <IWatchdog.h>

                         ^

compilation terminated.

Using library SPI at version 1.0 in folder: C:\Users\brian.ipsen\Documents\Arduino\hardware\STM32GENERIC-master\STM32\libraries\SPI 
exit status 1
Error compiling for board BLACK F407VG/VE/ZE/ZG boards.

ktand, 2018-10-26:
I think the problem is that ARDUINO_ARCH_STM32 is defined causing the files
in the generic HAL_STM32 to be compiled.

I have successfully compiled STM32F4 using PlatformIO with STM32GENERIC as
the platformio configuration in Marlin just includes the HAL_STM32F4.

Maybe changing:

#ifdef ARDUINO_ARCH_STM32

to

#if defined(ARDUINO_ARCH_STM32) && !defined (STM32GENERIC)

in the files in the HAL_STM32 will help.

bipsendk, 2018-10-26:
Managed to compile Marlin 2.0 (no changes to Configuration.h except to set board as BOARD_STM32F4 and SERIAL_PORT 1) with the Arduino_Core_STM32. Need to install from git since the version from package manager is not the latest and does not have support for the black F407VE. Instructions to install from git:
https://github.com/stm32duino/wiki/wiki/Using-git-repository

closed 2018-10-27 by bipsendk