MarlinFirmware/Marlin #12228: Z axis minimum stepper output 0.015 mm (6 Steps), Bug?

dahuafschmied, 2018-10-26:
Let's start with the given things:
My Z axis has a resolution of 400 steps / mm
1 Step ^= 0.0025 mm

how to reproduce:
i attached an oscilloscope to the "step" pin of the stepper driver to visualize the steps.
G91
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> 6 steps
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> no steps
G1 Z0.0025 F360 -> 6 steps
aso....

next:
G1 Z0.015 F360 -> 6 steps
G1 Z0.014 F360 -> no steps
G1 Z0.001 F360 -> 6 steps

Question:
Is this a wanted behavior?
Is it not possible to make single stepps?

thank you for clarification

[image]

gloomyandy, 2018-10-26:
Pretty sure this is just a consequence of the MIN_STEPS_PER_SEGMENT setting:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h#L915

Before you change it you may want to search the issues for discussions of why it is set to this value.

dahuafschmied, 2018-10-26:
ok. thats it. thanks.
i am waiting for #8606 to be implemented. 
i would like to see this disabled for mash bed leveling.

closed 2018-10-26 by dahuafschmied