MarlinFirmware/Marlin #12234: Error in Marlin 2.0 when using STM32F4 and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

bipsendk, 2018-10-27:
### Description

Error reported during build of STM32F4 configuration with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER defined
### Steps to Reproduce

Set up build environment
Enable REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER in configuration.h
Compile

**Expected behavior:** [What you expect to happen]
Successfull compile

**Actual behavior:** [What actually happens]
Error thrown:

C:\Users\user\AppData\Local\Temp\arduino_build_295107\sketch\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:88:4: error: #error "No valid condition for delays in 'ultralcd_st7920_u8glib_rrd_AVR.h'"

   #error "No valid condition for delays in 'ultralcd_st7920_u8glib_rrd_AVR.h'"

    ^~~~~

Command spawned to compile the file:

"C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\arm-none-eabi-gcc\\6-2017-q2-update/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:\\Users\\BRIAN~1.IPS\\AppData\\Local\\Temp\\arduino_build_295107/sketch/build_opt.h" -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD "-IS:\\3D-A8\\Firmware\\Marlin-bugfix-2.0.x (20181026)\\Marlin" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\cores\\arduino/avr" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\cores\\arduino/stm32" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\system/Drivers/STM32F4xx_HAL_Driver/Inc/" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\system/Drivers/STM32F4xx_HAL_Driver/Src/" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\system/STM32F4xx/" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\variants\\BLACK_F407VE/usb" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -DSTM32F4xx -DARDUINO=10900 -DARDUINO_BLACK_F407VE -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLACK_F407VE\"" -DSTM32F407xx -DHAL_UART_MODULE_ENABLED "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.3.0/CMSIS/Core/Include/" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\cores\\arduino" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\variants\\BLACK_F407VE" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\libraries\\SPI\\src" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.3.0\\libraries\\IWatchdog\\src" "-IC:\\Users\\user\\Documents\\Arduino\\libraries\\U8glib\\src" "C:\\Users\\BRIAN~1.IPS\\AppData\\Local\\Temp\\arduino_build_295107\\sketch\\src\\feature\\dac\\dac_mcp4728.cpp" -o "C:\\Users\\BRIAN~1.IPS\\AppData\\Local\\Temp\\arduino_build_295107\\sketch\\src\\feature\\dac\\dac_mcp4728.cpp.o"

It looks as if the error is triggered by that apparently no F_CPU has been defined, the MB definition not has been included in the list of if/defines - and the check is made even if ST7920_DELAY_1/2/3 already have been defined.

ktand, 2018-10-27:
I think It's because there is no delays defined for the board you are using in
ultralcd_st7920_u8glib_rrd_AVR.cpp

Instead of checking for different STM32 boards:

#elif MB(REMRAM_V1)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
#elif MB(ARMED)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)

I think a better solution would be define the delays per STM32 CPU, like
this:

#elif defined (STM32F4xx)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
#elif defined (STM32F7xx)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
#elif ...

What do you say, @thinkyhead and @hasenbanck?

hasenbanck, 2018-10-27:
Yeah, that might be better. Radical idea: Why not just use #elif defined (ARDUINO_ARCH_STM32)? Then we could later add exceptions to the rule when needed.

ktand, 2018-10-27:
Yes, @hasenbanck, that's better. I will provide a PR for this.

closed 2018-11-07 by thinkyhead