MarlinFirmware/Marlin #12235: Error in Marlin 2.0 - endstops - error: missing binary operator before token "("

TomTom076, 2018-10-27:
I used latest Marlin 2.0 fw with ATOM for a MKS SBASE 1.3. Build failed!
See error message in file.
Please HELP!
Thanks!
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2521610/Config.zip)
[image]

Ludy87, 2018-10-27:
Update your Version https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

TomTom076, 2018-10-27:
thanks.
i tried it with the link and the new fw, but nothing changed, error again.
Any ideas?

thinkyhead, 2018-10-29:
The only error I get with your configs and the current `bugfix-2.0.x` is this:
[image]

boelle, 2019-02-21:
@TomTom076 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-15 by p3p