MarlinFirmware/Marlin #12236: Gcode command "M420 Z" I need an explanation

marcodisco123, 2018-10-27:
Hi Everybody!
Since a while I installed the BLTouch sensor and I´m really sattisfied.
After the tunning I use the "M420 S1" command to use the grid and works.
I got that command spreads the leveling in the all height, for example on 10mm height a bit on the first layer and then the second, the third and so on till the end.

But what does the "M420 Z" command?
from the guide:

[Z<linear>] "Leveling fade Z height"

There are not so many examples about it and I got that has to be used with millimeters 
ex: "M420 Z1" means one millimeter
My interpretation is that it levels in 1 millimter instead the all height.
Is it like that?
Any experience?

Thanks in advance

marcodisco123, 2018-10-27:
hi!
Thanks for the reply!
Can you please explain better? (I´m a newbie)

marcodisco123, 2018-10-27:
ah ok!
So I didn´t get it at all... 
thanks a lot!

kAdonis, 2018-10-27:
you create a mesh so that the nozzle follows the not perfectly flat bed
the applied correction gets smaller (Faded) every layer until the Leveling fade Z height is reached
okay?

closed 2018-11-02 by thinkyhead