MarlinFirmware/Marlin #12237: blinkm problems with reprapdiscount smart controller on GT2560 controller

colinm001, 2018-10-27:
Using Marlin 1.1.9

I am using OctoPrint on a Raspberry Pi 3B+, to Control a Geeetech I3x with a GT2560 controller board and reprapdiscount smart controller (20x4) LCD display with SD Card reader. I have set up the Raspberry Pi and Octoprint to control and RGB LED strip, using LED Strip Controller plugin. 

I am able to successfully change the LEDs using M150 coded via gcode in Cura or Slic3r, which proves that the setup works

However, if I try to implement blinkm in Marlin the LCD display displays chains of characters and no print information. 

The LCD not responsive to pushing the control knob, although the print completes and the LEDs change colour as per the PRINTER_EVENT_LED code, and Octoprint displays all relevant print information

I am able to print normally if I disable blinkm

I have not made any changes to any files other than configuration.h to set up the board, LCD, BLTouch probe and to disable the SD Card reader on the reprapdiscount smart controller
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/2521798/Configuration.h.txt)

kAdonis, 2018-10-27:
I dont understand what you want to do
you might confuse something
this is a BlinkM:
https://thingm.com/products/blinkm/
did you mean this?

colinm001, 2018-10-27:
I want to have Marlin send printer event codes (M150 codes) which can be be intercepted by the OctoPrint, LED Strip Controller plugin and then change the colours on an LED strip connected to the Raspberry Pi.

I found a reference to this online, which indicated that blinkm had to be enabled within Marlin, and have been going down this route.

As I don't have a BlinkM device, I have obviously been on a wild goose chase, and this is an issue of my own making and not a bug within Marlin

Which LED type would I need to enable to achieve what I want to do?

The LED strip is controlled by the Raspberry Pi, through an appropriate driver and power board, and not directly to pins on the GT2560.

thinkyhead, 2018-10-27:
Marlin receives `M150`; it doesn't send it. In response, depending on the types of LEDs configured, Marlin will either send data over an I2C connection or directly set RGBW digital pins. You'll need to post in the OctoPrint discussion group to figure out your issue. We don't have any knowledge of the "LED Strip Controller plugin" and this issue queue is not for support, but only for Marlin development-related issues.

closed 2018-10-27 by thinkyhead