MarlinFirmware/Marlin #12238: Trigorilla board with TMC2130 and marlin 1.1.8 - X and E0 drivers not detected

petr222, 2018-10-27:
Hi, I upgraded Trigorilla in my Anycubic Kossel plus with TMC2130 drivers.
Everything should be done regarding to common manual - removing R2
resistor on all drivers to enable SPI mode, bridge CFG4 and CFG5 and connect
all 4 pins to pins on motherboard. 

With driver monitoring disabled, printer boot up, but auto home command
move only Y and Z axis, X is still and there are some strange sounds coming
from printer. Extruder motor have same problem, no movement using axis control.

In pin_ramps, CS is configured to use following pins:

//
// Steppers
//
#define X_STEP_PIN         54
#define X_DIR_PIN          55
#define X_ENABLE_PIN       38
#define X_CS_PIN           4 //53

#define Y_STEP_PIN         60
#define Y_DIR_PIN          61
#define Y_ENABLE_PIN       56
#define Y_CS_PIN           5 //49

#define Z_STEP_PIN         46
#define Z_DIR_PIN          48
#define Z_ENABLE_PIN       62
#define Z_CS_PIN           6 //40

#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      11 //24
#define E0_CS_PIN          42

#define E1_STEP_PIN        36
#define E1_DIR_PIN         34
#define E1_ENABLE_PIN      30
#define E1_CS_PIN          44

If I understand configuration correctly, same pins are used by servos -
should it be disabled in configuration ? There are no servos connected,
just driver wires. 

With driver monitoring enabled, printer is killed when it is switched off
with following log:

Marlin 1.1.8
echo: Last Updated: 2017-12-25 12:00 | Author: (none, default config)
echo:Compiled: Oct 27 2018
echo: Free Memory: 2402  PlannerBufferBytes: 1232
echo:V47 stored settings retrieved (614 bytes; crc 28318)
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z80.00 E96.00
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z200.00 E50.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z3000 E3000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P2000.00 R2000.00 T2000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X5.00 Y5.00 Z5.00 E5.00
echo:Endstop adjustment:
echo:  M666 X0.00 Y-1.71 Z-4.55
echo:Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
echo:  M665 L267.00 R132.85 H302.92 S100.00 B93.00 X-0.02 Y-0.13 Z0.15
echo:Material heatup parameters:
echo:  M145 S0 H180 B50 F255
echo:  M145 S1 H240 B100 F255
echo:PID settings:
echo:  M301 P18.14 I1.33 D62.05
echo:  M304 P363.20 I71.51 D461.18
echo:Z-Probe Offset (mm):
echo:  M851 Z-17.21
echo:Stepper driver current:
echo:  M906 X 0 Y 0 Z 0 E0 0
echo:SD card ok
echo:SD card ok
X driver error detected:
		X	Y	Z	E0
Enabled		false	false	false	false
Set current	0	0	0	0
RMS current	1767	1767	1767	1767
MAX current	2491	2491	2491	2491
Run current	31/31	31/31	31/31	31/31
Hold current	31/31	31/31	31/31	31/31
CS actual		0/31	31/31	31/31	0/31
PWM scale	0	2	2	0
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	true	true	false
msteps		256	16	16	256
tstep		0	1048575	1048575	0
pwm
threshold		98	98	98	274
[mm/s]		1613.52	100.85	100.85	480.92
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time		0	5	5	0
blank time	16	24	24	16
hysterisis
-end		-3	2	2	-3
-start		1	3	3	1
Stallguard thrs	0	0	0	0
DRVSTATUS	X	Y	Z	E0
stallguard
sg_result		0	0	0	0
fsactive
stst			X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
	X = 0x00:00:00:00
	Y = 0x80:1F:00:00
	Z = 0x80:1F:00:00
	E0 = 0x00:00:00:00
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

If I understand it correctly, drivers for X and E0 did not
communicate with the board, as registers are all zeroes. 
Therefore only data from Y and Z drivers are actually 
from drivers.

I switched X / Y and Z / E0 drivers and error is the same,
therefore problem should be in Marlin configuration.

My main question is - could this problem be caused by
PIN configuration conflict - using same PINs for servo
operation and drivers control, for example ?  Should 
another #define entries using same PIN number be commented ?

As printer does not respond to M43 command, is there a way
to find which PIN is used for what ?

petr222, 2018-10-27:
Upgraded to Marlin 2.0 debug, configured everything again and same problem.

In 2.0 there is config for Trigorilla board, in 1.1.8 plain Ramps was used.

Remapped SERVO_1, SERVO_2, SERVO_3 and FIL_RUNOUT to unused ports
60+. Switched X driver with E0 driver and now X, Y and Z drivers are detected
and working. Therefore is seems to be hardware problem with one of 
TMC2130 drivers, as problem move to position where this driver is.

PINs 4, 5 and 6 for X, Y and Z axis are correct. I am unable to find correct
pin or port for E0 driver, tried PIN 11 and 21 (used by other users in their
config) but no driver is detected.

Just a suggestion - TMC2130 configuration using servo ports should
be included in official Marlin configuration, when there already is 
Trigorilla configuration file.

boelle, 2019-02-21:
@petr222 problem solved in latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-08-30 by shitcreek