MarlinFirmware/Marlin #12239: Y offset / bltouch issues MKS Sbase v1.3  LPC1768

gauix4, 2018-10-27:
brand new still this problem Marlin no compilation problem stepper well settled no problem after a while there is offset in Y...
100% sure it's a Marlin problem with other map everything works fine
1 - the BLtouch tends to deploy itself
2- the Y always shifts after a while

gauix4, 2018-10-27:
we see that there is a problem because the temperature varies by itself
[image]

p3p, 2018-10-27:
If you run `pio update` in the atom terminal what version of `framework-arduino-lpc176x` are you running, 

Always supply the information requested in the issue template, such as full configs. additional information about your hardware setup may also help us rather than just specifying that things that don't work.

gauix4, 2018-10-27:
[image]

gauix4, 2018-10-27:
[image]

gauix4, 2018-10-27:
@p3p 
that's what you want I have a little trouble with atom I did not know

gauix4, 2018-10-27:
[image]

p3p, 2018-10-27:
did `framework-arduino-lpc176x` update when you ran `pio update` or were you already on 0.0.5?

gauix4, 2018-10-27:
@p3p this one was not showing at all this is the first screen shot I made by sending the command

gauix4, 2018-10-27:
but I think that's it there were a lot of things that were not up to date that went on as 
the best way to know is that i re test ^^

p3p, 2018-10-27:
I know that's why I asked, it hadn't finished at the point you took the screenshot. If it did update it may have fixed your BLTouch issues as there was a known issue with it for a few days but if you are only trying this today it's unlikely platformio downloaded the old version.

Please supply your configuration files and hardware setup, including stepper drivers and temperature sensors.

gauix4, 2018-10-27:
@p3p Driver ES-D508 Y "closed-loop"
X TMC 2208 no serie connect

I am not at all a beginner configuration.h and well configure the probes also there is really that with this card that I have problems all the others have no problem the card is new I have changed two times to be sure but still the same problems so no external problem just the marlin or update problems as you showed me

gloomyandy, 2018-10-27:
To save you both sometime, here is a previous post about this problem (see the following posts for some followup):

gauix4, 2018-10-27:
[dossier sans titre.zip](https://github.com/MarlinFirmware/Marlin/files/2521978/dossier.sans.titre.zip)

gauix4, 2018-10-27:
@sombre I forgot to add the link ^^

p3p, 2018-10-27:
Your stepper driver is not in general use so we do not have a default configuration for it, you need to set the values in `configuration_adv.h` to match the datasheet.

Looking at http://www.leadshine.com/UploadFile/Down/ES-D508d_V1.2.pdf (best guess on my behalf at your driver) it appears your driver needs extra long pulses compared to the usual divers (2.5us minimum), so you will need to set `MINIMUM_STEPPER_PULSE` to an appropriate value, remember 2.5us is the minimum so 3 or even 4 should guarantee you driver picks it up within a margin for error. 

I assume whatever bus driver (logic level converter) you are using is suitable for the slew and frequency requirements.

gauix4, 2018-10-27:
@p3p I did as I told you I'm not a beginner all is good

p3p, 2018-10-27:
According to your config you did not, your setup for A4988 (the default drivers), that work down to a few 100 nano second pulse width.

gauix4, 2018-10-27:
@p3p I stay with the update I tell you that in 20 minutes the time that everything starts on the road

gauix4, 2018-10-27:
@p3p nothing has changed always shift in Y always the bltouch that does anything

gauix4, 2018-10-27:
@p3p I feel that because I have large dimensions
put small dimensions everything works properly 300X300X300

gauix4, 2018-10-27:
https://youtu.be/dWgYPQUOQVc

gauix4, 2018-10-27:
@p3p a compilation problem nothing but still the same problems even with a delay add

gloomyandy, 2018-10-27:
@guanix Have you used that driver successfully with any other board and/or firmware? If so which? Also have you used the level shifter that you built successfully with anything?

gauix4, 2018-10-27:
yes
marlin 2xx no
marlin 1.xx ok
MKS Gens 1.4 ok
lerdge S ok
lerde X ok
mks robin ok

gauix4, 2018-10-27:
that's why from the beginning I say it's 100% sure it's the marlin I never had any other worries before except the passage of Marlin 2

p3p, 2018-10-27:
What was fixed by reducing the dimensions?

We are talking about a lot of separate issues here, which is a bit confusing with you short responses, this is for the missing step on your Y axis, you are now running Marlin on a much more powerful Controller, signals that used to be slow enough for your driver because it was as fast as the AVR could produce them could now be too fast. Have you read the datasheet and made sure all the settings are correct? such as `MINIMUM_STEPPER_DIR_DELAY` that defaults to 50ns.. the datasheet specifies this needs to a much larger value of 5000ns,

I'm not saying there is nothing wrong with Marlin, but there a numerous people running Marlin on the MKS Sbase so it being a general issue is unlikely we are just trying to narrow the problem down for you.

gauix4, 2018-10-27:
I just reinstalled the gen 1.4 everything works properly...

gauix4, 2018-10-27:
last week I used the smoothieware software everything worked...

gloomyandy, 2018-10-27:
Are you running Marlin 2.0.x on the gen 1.4? If not perhaps you can try it. Does the BL touch work on the gen 1.4 with the full bed size defined? 

Why are you trying to use Marlin rather then Smoothieware?

gauix4, 2018-10-27:
@gloomyandy  I'll have to try yes because I'm on MARLIN 1 and everything works perfectly
because we can do nothing at all with smoothie firmware
il me faut un auto bed leveling car mon imprimante est trop grosse impossible d'imprimer sans bed auto leveling  smoothie has one but it is very poorly managed

gauix4, 2018-10-28:
good marlin 1 on the gen 1. ' it does not work either it shifts Y but I've already tried it marchaigood marlin 1 on the gen 1. ' it does not work either it shifts Y but I've already tried it marchai 
I do not understand anymore

gauix4, 2018-10-28:
test that it does not work either
check my signal 0/5 volts ok
I use booster at the
https://fr.aliexpress.com/item/LERDGE-3D-Printer-Board-Adapter-Module-External-High-Power-Switching-Module-for-Microstep-Driver-with-3D/32821023876.html

[image]

gauix4, 2018-10-28:
person to an idea ?

p3p, 2018-10-28:
Sorry Simon at this point without the hardware to debug with, or anyway to reproduce the problems you are having, it's difficult to help.

That stepper configuration should have more than sufficient delays for your Y stepper driver, the problem with levelling at large dimensions worries me but I have no way to test that and it's not really my area, you may have better luck with UBL rather than bilinear. 

The temperature fluctuations shouldn't be possible without huge amounts of noise as there is a lot of software filtering, but that at least I can test.

gauix4, 2018-10-29:
I can still try anyway I am open to all at this stage I do not know what to do other than change its engine more I never liked them

gauix4, 2018-10-29:
[image]
[image]
[image]

gauix4, 2018-10-29:
@p3p we can see that the shift is not random

gauix4, 2018-10-29:
@p3p this is on the last 1xx Marlin it does the same thing on the 2xx marlin 
last time i tried with a former sailor one that worked fine so i think there is an update made that generates this problem when do you think?

gloomyandy, 2018-10-29:
If you think that is the case, work your way back through versions to see if you can pinpoint which change is causing the problem. Have you made any changes to your configuration since you had a working system? What about the hardware any changes to that? Are you using the same speeds wtc. as before?

gauix4, 2018-11-01:
it works according to the parts there are parts it prints across there are non parts

gauix4, 2018-11-01:
and therefore order real brushless motor (clearPath ref: CPM-SDSK-2310S-EQN) even though I think the problem will still be there

gauix4, 2018-11-22:
I received my new clearPath engine
and nothing function so I started my research
and I realized that the problem was coming from ENABLE
the signal is not clear at all when I plug on 3.3V everything works
[image]

closed 2018-11-26 by gauix4

emretrk, 2018-12-04:
Hi everyone,

I am a student in Turkey and ı also see like that problem. My drivers are also leadshine es-d508 and es-d808. I couldn't solve this problem. Are there anyone to solve this problem ? 

3D printer is mu capstone project. ı must solve this...