MarlinFirmware/Marlin #12241: features requres : add progress parameter to M150 ?

fred2088, 2018-10-27:
Dear Sirs,

would it possible to add "L" progress parameter to M150 ?

   *   M150 L0~100     ; Line Progress 0%~100%

------------------------------------------------------ all this works from 1.1.x using fastled
if (parser.seen('L')) 
   LedstripMeter( parser.seen('L') ? (int)parser.value_ushort() : 0 ) ;

------------------------ printing from sd card

  lcd_implementation_status_screen();
#if HAS_COLOR_LEDS
  if (IS_SD_PRINTING) 
      LedstripMeter(card.percentDone());
#endif

void LedstripMeter(int percent) {
     int i = percent * 0.01 * LEDSTRIP_NLED -1;
     int Hue = 50;
     if (percent > 0)
//        leds[i] = CHSV(Hue, 0, 255);
        leds[i] = CRGB(255, 0, 0);
     FastLED.show();
}

fred2088, 2018-10-27:
added "L" progress parameter to M150 for 2.0.x for my own use, please enchance if someone feel it's useful

----------------------------------------------   added following code to leds.h 
  static void setup(); // init()
  static void meter(int percent);

----------------------------------------------   added following code to leds.cpp 

void LEDLights::meter(int percent) {
  #if ENABLED(NEOPIXEL_LED)
     int i = percent * 0.01 * NEOPIXEL_PIXELS -1;
     if (percent > 0)
      pixels.setPixelColor(i, pixels.Color(255, 0, 0));
      pixels.show();
  #endif
}
----------------------------------------------   added following code to M150.cpp 

 *   M150 L0~100     ; Line Progress 0%~100%
 *   M150 P          ; Set LED full brightness
 */
void GcodeSuite::M150() {
if (parser.seen('L')) 
   leds.meter( parser.seen('L') ? (int)parser.value_ushort() : 0 ) ;
else
	leds.set_color(MakeLEDColor(
    parser.seen('R') ? (parser.has_value() ? parser.value_byte() : 255) : 0,
    parser.seen('U') ? (parser.has_value() ? parser.value_byte() : 255) : 0,
    parser.seen('B') ? (parser.has_value() ? parser.value_byte() : 255) : 0,
    parser.seen('W') ? (parser.has_value() ? parser.value_byte() : 255) : 0,
    parser.seen('P') ? (parser.has_value() ? parser.value_byte() : 255) : pixels.getBrightness()
  ));
}

sjasonsmith, 2020-11-10:
I think this is an interesting feature request. The parameter doesn't show up in the RepRap wiki, so it would be something new. The RepRap wiki lists an "S" option which sets a particular number of LEDs. The percent option would certainly be more flexible.

Tannoo, 2023-05-08:
This would be a good addition if you could select which strip or led port to use for it.

fred2088, 2023-05-08:
> This would be a good addition if you could select which strip or led port to use for it.

it's already done from 2.1.x 

 * With NEOPIXEL_LED:
 *  I<index>  Set the NeoPixel index to affect. Default: All
 *  K         Keep all unspecified values unchanged instead of setting to 0.
 *
 * With NEOPIXEL2_SEPARATE:
 *  S<index>  The NeoPixel strip to set. Default: All.

Tannoo, 2023-05-08:
I am sorry. I didn't see that this was an old post.