MarlinFirmware/Marlin #12244: H-Bot/CoreXY axis direction issues

FlailAway, 2018-10-27:
I am building an H-Bot and have configured marlin 1.1.9 as such. I can get the Alpha(X) and Beta(Y) axis to manually move (Prepare/Move Axis) in the correct directions but then Homing goes bananas.

Homing switches at front/Left.
G28 X0 Homes the Alpha correctly
G28 Y0 Homes the Beta correctly

Using G28 will first Home Alpha and then try Beta. Instead of the Beta going to the switch, it goes away from it in the "+" direction. It hits the end "+" of travel, rattles for a second or two then Homes correctly.

Can anyone/someone please point me at somewhere that shows the Configuration.h set up for H-Bot, as clearly, the basic CoreXY settings are not cutting it. I have tried all the connector swap combinations and nothing is improving it.

Thanks

rjacobs1969, 2018-10-31:
I'm building a mini coreXY printer and had some problems with directions and homing. But after some try and error I found the correct settings. 

When X/Y is moving in the correct directions but homing isn´t check out the 
X_HOME_DIR and Y_HOME_DIR settings
Also check X_MIN_ENDSTOP_INVERTING / Y_MIN_ENDSTOP_INVERTING having these wrong also give strange homing behaviour

towe96, 2018-10-31:
According to my short research, H-Bot has one axis inverted compared to CoreXY.
The homing direction settings comes to mind, or perhaps the CoreYX motion setting.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-16 by thinkyhead