MarlinFirmware/Marlin #12248: TMC2130 Sensorless Homming feels rougher after update to v1.1.9

FelixHauser, 2018-10-27:
### Description

Sensorless Homing with Marlin 1.1.9 feels a lot rougher as it was with Marlin 1.1.8

### Steps to Reproduce

Update to Marlin 1.1.9 with the equivalent setting as with 1.1.8

**Expected behavior:** 

I expected sensorless homing to be as quiet as it was before.

**Actual behavior:** 

It's not. It feels like the motors are ramming to the MIN positions for the homing to trigger. It's quite a lot noisier, evidently.

#### Additional Information

So I've been finally able to tackle down on Marlin 1.1.9. My printer is a cartesian prusa-like with a steel frame. Out of the 5 printers I own, this is the one delivering the nicest parts. The controller board is a Rumba v1.1 of which I'm pretty happy about. The X and Y axis are driven by 2 original TMC2130s, connected to the correct ports. The Z and E axis were driven by 2 original TMC2208 on Stand Alone Mode. Everything was working really well, but I wanted to take advantage of the new features of v1.1.9  
I copy-pasted the settings of v1.1.8: steps/mm, federates, accelerations etc. I also copied (or transferred) the settings for the TMC drivers. I ended up changing the Extruder driver in order to avoid the problem mentioned on issue #11825 but everything else was ok. I configured as Software-SPI with the same pins I used before (MISO:50; MOSI:51 and SCK:52) ans so far so good.
The only problem I'm seeing is, as mentioned, that the Sensorless Homing seems way rougher compared as it was before. With 1.1.8 seemed much smoother. I tried to dial the sensitivity down to 5, but it doesn't help all that much. Lower than 5 leads to false positives. Also, I disabled the homing bump as suggested. Talking to a printing buddy of mine, he experienced the same behavior using the same board and drivers.

Is this behavior normal or is there something I missed to configure? Is somebody else experiencing this?. I haven't seen any issue regarding this topic, if I missed I apologize in advance.

Thanks in advance.

[Configuration_1.1.8.txt](https://github.com/MarlinFirmware/Marlin/files/2522291/Configuration_1.1.8.txt)
[Configuration_1.1.9.txt](https://github.com/MarlinFirmware/Marlin/files/2522292/Configuration_1.1.9.txt)
[Configuration_adv_1.1.8.txt](https://github.com/MarlinFirmware/Marlin/files/2522293/Configuration_adv_1.1.8.txt)
[Configuration_adv_1.1.9.txt](https://github.com/MarlinFirmware/Marlin/files/2522294/Configuration_adv_1.1.9.txt)
[pins_RUMBA_1.1.8.txt](https://github.com/MarlinFirmware/Marlin/files/2522295/pins_RUMBA_1.1.8.txt)
[pins_RUMBA_1.1.9.txt](https://github.com/MarlinFirmware/Marlin/files/2522296/pins_RUMBA_1.1.9.txt)

thinkyhead, 2018-11-02:
@teemuatlut — Any other stepper driver tuning that Felix might try to get less "rough" homing?

Gabbajoe, 2018-12-08:
Hi,
I experience the same issue, with Marlin 1.1.8 the homing is quite and smooth. After updatet to Marlin 1.1.x the homing of the X axis is making noise and feels bad, for the Y axis it is as before.
I use a MKS GEN_L Board with 4 original TMC2130 drivers and all connect "default" no special modification. 
I found a solution that works fine for me. I inverted the direction of my X axis stepper motor via Configuration.h after I changed it to not invert the direction the X axis homing is again quite and smooth. So I have to revers now my connector cable.

Maybe this helps to debug the issues.

Happy printing,
Gabbajoe

worlderic, 2018-12-22:
sensorless homing need “stall guard”，this only work in tmc2130‘s coolstep mode，&must turn off stealchop，so noise is larger

boelle, 2019-02-21:
@FelixHauser problem solved in latest bugfix 2.0?

FelixHauser, 2019-02-24:
@boelle  I´m trying to compile Marlin 2.0 on a 32bits board I have. I will try it if I´m successful on that front

boelle, 2019-02-24:
oki, you dont have to compile it for a 32 bit board, 2.0 is both 8 bit and 32 bit

boelle, 2019-03-06:
@FelixHauser any success?

boelle, 2019-03-13:
@thinkyhead think we can close this one

FelixHauser, 2019-03-13:
are you in a hurry for something? Sincerely, I have the impression of being stalked upon. You show up in every single issue I open...

I will try to test it over the weekend and report back a soon as I can. I will try @Gabbajoe 's suggestion first, though.

boelle, 2019-03-13:
> You show up in every single issue I open...

and you are not the only one opening issue's, i give everyone the same care

thinkyhead, 2019-03-14:
Please keep conversation to technical subjects.

shitcreek, 2019-05-14:
@FelixHauser Were you able to resolve this issue?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-09-24 by shitcreek