MarlinFirmware/Marlin #12250: Linear Advance 1.5 Causing Print To Stop

inventabuild, 2018-10-28:
I'm using Marlin 1.1.9 with LA 1.5 and it's causing my prints to pause mid-print: https://youtu.be/EYoc8PGj3Tw

Any idea what causes this?

Sebastianv650, 2018-10-28:
Is this repeatable at this specific point in the gcode? If so, try to isolate the corresponding gcode lines and also try them with K=0 to see if that changes anything. Then please post your config files together with the gcode sniplet.

swilkens, 2018-10-28:
Are you using USB / Serial connection to print? (i.e. from PC, via octoprint or similar)

If so, the buffer may simply be running out or you are experiencing communication issues. If this is the case, place the file on an SD card and attempt the print again.

thinkyhead, 2018-10-28:
Do you happen to be using Simplify3D as your slicer? There's a known issue where it will add very small and very slow retractions that make the printer seem like it's pausing.

boelle, 2019-02-21:
@inventabuild problem solved with latest bugfix 2.0 ?

inventabuild, 2019-02-22:
Simplify 3D v4.1.1 appears to have fixed the problem.

boelle, 2019-02-22:
then click close below

closed 2019-02-22 by inventabuild