MarlinFirmware/Marlin #12251: sbase Marlin 2.0 Error: Detected unknown package 'framework-arduino-lpc176x'

keshantao, 2018-10-28:
I was trying to compile Marlin 2.0 for my sbase tonight, and I always get 

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x
/archive/master.zip; board: nxp_lpc1768; framework: arduino)
-------------------------------------------------------------------------------
-
PackageManager: Installing framework-arduino-lpc176x @ >=0.0.2
Error: Detected unknown package 'framework-arduino-lpc176x'

I made change to the platformio.ini file as the env_default= LPC1768
and all necessary change in Configuration.h
with/without a sbase board connected.

Have no success, :-(, advice is really appreciated.

gauix4, 2018-10-28:
If you run pio update in the atom terminal what version of framework-arduino-lpc176x are you running,

keshantao, 2018-10-28:
it compiles now, I suspect my pfsense blocked something.

InsanityAutomation, 2018-10-28:
Must've forgotten to sacrifice an old freebsd beastie before downloading technical things....

gauix4, 2018-10-28:
so you can close this request Thank you

closed 2018-10-28 by keshantao