MarlinFirmware/Marlin #12253: "Error: Hotend too cold" on connection to the printer

flcu, 2018-10-28:
### Description

Receving "Hotend too cold" Error Message on connecting to the printer in Replicator-GUI

### Steps to Reproduce

I simply connect to my printer, then the error message appears. Connect Log from replicator attached.

**Expected behavior:** Connecting to the printer without an error message

**Actual behavior:** I get an annoying replicator alert on connect

I'm using a mixing extruder with 2 extruders and one hotend. I've traced the error message to the tool_change.cpp and I'm suspecting that somehow a tool change is triggered/attemped at connect?

The printer works fine afterwards, as the temperature reading and everything works fine?

I'm using Marlin Bugfix-2.0.x from 28.10.2018

#### Additional Information
[20181028_err_log_connect_too_cold.log](https://github.com/MarlinFirmware/Marlin/files/2522608/20181028_err_log_connect_too_cold.log)
[20181028_config_error_too_cold.zip](https://github.com/MarlinFirmware/Marlin/files/2522606/20181028_config_error_too_cold.zip)

thinkyhead, 2018-10-28:
@InsanityAutomation — This machine tries to do a tool-change on startup. And it's a `SINGLENOZZLE`. So it's invoking the filament swap and throwing an error. Re-check the tool-change code for `SINGLENOZZLE` to see whether it's trying to invoke a filament swap when it shouldn't be. Clearly there are occasions when users will want to do a tool-change with a `SINGLENOZZLE` setup but not swap filaments.

thinkyhead, 2018-10-28:
Meanwhile, @flcu — Change your `SINGLENOZZLE_SWAP_LENGTH` to 0.

flcu, 2018-10-28:
@thinkyhead: this solves at least the error message - as I'm not using any toolchange functionality at the moment not an issue for me.

InsanityAutomation, 2018-10-28:
Good morning, I'll take a closer look shortly, but in the meantime can you try the updated set of code at https://github.com/MarlinFirmware/Marlin/pull/12137

thinkyhead, 2018-11-02:
You can close this if the original issue is solved.

boelle, 2019-03-13:
@thinkyhead think we can close this one

flcu, 2019-03-13:
Sorry guys, I've since switched hardware and software. The proposed fix "Change your SINGLENOZZLE_SWAP_LENGTH to 0." worked even though it's probabely not correct (as the length for a swap is different)...

closed 2019-03-13 by flcu