MarlinFirmware/Marlin #12254: G29 turns off heat to nozzle and bed.

bugilt, 2018-10-28:
When I use G29 in my starting script the heat turns off on both the nozzle and bed. 

Here is what the terminal shows.

```
Recv: 0 1 2
Recv: 0 -0.018 -0.043 +0.054
Recv: 1 -0.029 -0.010 +0.018
Recv: 2 -0.083 -0.028 +0.027
Recv: 
Recv: X:298.00 Y:260.00 Z:10.68 E:2.65 Count X:23840 Y:20800 Z:4260
Recv: ok
Send: N11 M400*23
Recv: ok
Send: N12 M114*20
Recv: X:298.00 Y:260.00 Z:10.68 E:2.65 Count X:23840 Y:20800 Z:4260
Recv: ok
Send: N13 M140 S0*87
Recv: ok
Send: N14 M104 S0*80
Recv: ok
Send: N15 G1 Z0.2 F3000*47
Recv: ok
Send: N16 G92 E0*112
Recv: X:298.00 Y:260.00 Z:0.20 E:0.00 Count X:23840 Y:20800 Z:4260
Recv: ok
Send: N17 G1 X160 E15 F600*96
Recv: T:207.62 /0.00 B:65.08 /0.00 @:0 B@:0
Recv: T:207.23 /0.00 B:64.94 /0.00 @:0 B@:0
Recv: T:206.91 /0.00 B:64.92 /0.00 @:0 B@:0
Recv: T:206.33 /0.00 B:64.72 /0.00 @:0 B@:0
Recv: ok
Send: N18 M83*33
Recv: ok
Send: N19 G1 E-6.50000 F5100*23
Recv: ok
Send: N20 G91*35
Recv: ok
Send: N21 G1 Z0.200 F1000*42
Recv: ok
Send: N22 G90*32
Recv: ok
```

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/2522611/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/2522612/Configuration_adv.h.txt)

thinkyhead, 2018-10-28:
Looks like your host is turning off your heaters.
```
Send: N13 M140 S0*87
Send: N14 M104 S0*80
```

bugilt, 2018-10-28:
Now to figure out why.

bugilt, 2018-10-28:
I guess octopi is the culprit.