MarlinFirmware/Marlin #12260: Second probing speed for Z is the same as the first probing (G28)

achimu, 2018-10-28:
Hello,

I changed the version from 1.1.8 to 1.1.9. With this change, the second probing speed for Z at G28 is the same as the first probing. My settings are:

#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)
// #define MULTIPLE_PROBING 2
#define HOMING_FEEDRATE_Z  (4*60)

the Z probing at the G28 do always 2 probes, even if the MULTIPLE_PROBING is commented out. And with MULTIPLE_PROBING 2 the Z probing also is the same speed from first to second probe. Are there any changes between 1.1.8 and 1.1.9 in configuration that I have overlooked?

Achim

swilkens, 2018-10-31:
The second homing (the bump) speed is configured by `HOMING_BUMP_DIVISOR` in `Configuration_adv.h`. What does that show for you:

> // Homing hits each endstop, retracts by these distances, then does a slower bump.
> #define X_HOME_BUMP_MM 5
> #define Y_HOME_BUMP_MM 5
> #define Z_HOME_BUMP_MM 2
> #define HOMING_BUMP_DIVISOR { 2, 2, 4 }  // Re-Bump Speed Divisor (Divides the Homing Feedrate)
> #define QUICK_HOME                       // If homing includes X and Y, do a diagonal move initially

AnHardt, 2018-10-31:
If both of this feedrates are above DEFAULT_MAX_FEEDRATE for z you will also se no difference because they are limited to that.

achimu, 2018-10-31:
After a lot of testing and changing parameters, I found (my ?) problem. First, with BLTOUCH the following parameters are used (not the "BUMP" parameters)

#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)
#define HOMING_FEEDRATE_Z (4*60)

I found out, that the dividing (/2) is not working correct. With
#define Z_PROBE_SPEED_FAST 240
#define Z_PROBE_SPEED_SLOW 120
15:45:45.507 : >>> do_homing_move(Z, 5.00, 4.00)
15:45:48.569 : >>> do_homing_move(Z, -10.00, 2.00)
I get teh same speed for the second probing, with
#define Z_PROBE_SPEED_FAST 240
#define Z_PROBE_SPEED_SLOW 60
16:12:46.384 : >>> do_homing_move(Z, 5.00, 4.00)
16:12:49.462 : >>> do_homing_move(Z, -10.00, 1.00)
I get the half speed for the second probing, with
#define Z_PROBE_SPEED_SLOW 30
I get the quarter for the second probing. 
-> therfore I have no debug output

I measured this by increasing the parameter
#define Z_CLEARANCE_BETWEEN_PROBES  20 // Z Clearance between probe points
 and stop the time for moving the nozzle down.

Is there another parameter that I overlooked?

achimu, 2018-11-01:
I found that reason for my problem.
--> M502 
After loading the firmware with new settings, the 
DEFAULT_MAX_FEEDRATE          { 400, 400, 4, 45 }
is not updated without M502. Every time the "old" EEPROM settings are used. Beginners mistake.........

closed 2018-11-01 by achimu