MarlinFirmware/Marlin #12262: Bugfix 2.x Compile Error

landodragon141, 2018-10-28:
Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
SYSTEM: ATMEGA2560 16MHz 8KB RAM (248KB Flash)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.2.2 #cefefbe
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
|-- <Adafruit NeoPixel> 1.1.3
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.6.2
|   |-- <SPI> 1.0
|-- <c1921b4>
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0
|-- <Firmata> 2.5.6
|   |-- <SoftwareSerial> 1.0
|-- <Servo> 1.1.2
|   |-- <Firmata> 2.5.6
|   |   |-- <SoftwareSerial> 1.0
|-- <WiFi> 1.2.7
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\HAL.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\HAL_spi_AVR.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\MarlinSerial.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\persistent_store_eeprom.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\servo_AVR.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\watchdog_AVR.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\I2cEeprom.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\SpiEeprom.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\backtrace\unwarm.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\backtrace\unwarm_arm.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\backtrace\unwarm_thumb.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\backtrace\unwarmbytab.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\backtrace\unwarmmem.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\backtrace\unwinder.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\persistent_store_api.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\Marlin.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\core\serial.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\core\utility.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\I2CPositionEncoder.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\Max7219_Debug_LEDs.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\baricuda.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\bedlevel\abl\abl.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\bedlevel\ubl\ubl.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\bedlevel\ubl\ubl_G29.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\bedlevel\ubl\ubl_motion.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\caselight.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\closedloop.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\controllerfan.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\dac\dac_dac084s085.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\feature\dac\dac_mcp4728.cpp.o
[... 448 lines cut ...]
============================================================================================================================================== [SUMMARY] ==============================================================================================================================================
Environment megaatmega2560              [ERROR]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [SKIP]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
===================================================================================================================================== [ERROR] Took 90.17 seconds =====================================================================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

gloomyandy, 2018-10-28:
It would help if you provide the requested files (your config) along with the other items requested in the issue template.

Are you trying to use a TMC2208 stepper driver (or other TMC driver) that uses a serial interface to control it? Do you have ENDSTOP_INTERRUPTS_FEATURE enabled? What happens if you disable this option?

landodragon141, 2018-10-28:
Here's a link to my config files
https://1drv.ms/u/s!ApeqTrXyrCKQgoZ7v39GizUTU0R9vw

landodragon141, 2018-10-29:
I use tmc2130s with sensorless homing. It runs on the hw spi bus. I also use end stop interrupts. I've compiled this same configuration in the past without this error.

thinkyhead, 2018-10-29:
We can't see your config at that archaic file server link. You should have made a ZIP file and dropped it on your original post, or on a reply. Anyway, two things:

**One:** See if it helps to disable `ENDSTOP_INTERRUPTS`.

**Two:** This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2018-10-29 by thinkyhead

p3p, 2018-10-29:
Can't blame thinkyhead for being a little cranky, the vast majority of issues Marlin gets are user support, and usually badly configured printers that a quick google search would have solved. In your case you should have searched the issues for`"SoftwareSerial.cpp.o (symbol from plugin): In function 'SoftwareSerial::read()'` you would have seen this issue has already come up, it is a bug, well a known limitiation, but not in Marlin as such.

The update and consolidation of TMC support into the TMCStepper the library has made the `ENDSTOP_INTERRUPTS` feature incompatible with TMC drivers, until a workaround can be found.

gloomyandy, 2018-10-29:
As above if you seach on the suggested terms you will find the following which gives a lot more details of the problem....
https://github.com/MarlinFirmware/Marlin/issues/12078

landodragon141, 2018-10-29:
Hmm it looks like there is a possible work around but it would require using a library called AltSoftSerial as it is capable of coexisting with interrupts. I'm of course completely ignorant to the feasibility of doing this. I'm a great system admin but a mediocre programmer especially when it comes to the very sensitive highly optimized nature of marlin. Kudos to you guys that keep this project going!