MarlinFirmware/Marlin #12263: Marlin 2 compiled, flashed, but just won't boot up

keshantao, 2018-10-29:
I finally got Marlin 2.0 compiled without any errors, now a new problem came up and I fought it this afternoon with no success - my sbase just won't boot. (or booted only once for the last 5 hours I tried)

I first compiled a firmware with all my configuration modified to my needs, and put that to a 125M or 16G MicroSD card (I tried both), and power up the board with USB or 24V power or both (I tried all), and I hit the reset button.  The firmware.bin did change to FIRMWARE file, and that's it, no further response.

after this happened, even if I put back the sd card with working smoothie firmware, the board still won't boot, I have to drag a smoothie firmware.bin on the sd card, and power up the board, it changed back to smoothie.

then I compiled a more plain firmware with minimal change to the configuration (with only TMC2208, enabled GLCD and changed board type), and to my surprise it worked.  

So I again changed back to the costumed firmware, and from that moment, I didn't boot up the machine no matter which Marlin firmware I put on my SD card, every time the firmware.bin was flashed (name changed to FIRMWARE) but just won't boot, what did I kept doing wrong?

[firmware(plain).zip](https://github.com/MarlinFirmware/Marlin/files/2523402/firmware.plain.zip)
[firmware(customized).zip](https://github.com/MarlinFirmware/Marlin/files/2523403/firmware.customized.zip)

thinkyhead, 2018-10-29:
Marlin 2.0 is still under heavy development and not all features are supported. Start with a minimal configuration and only enable one feature at a time until it becomes clear which feature or features are causing your problem.

p3p, 2018-10-29:
What do you mean by it wont boot?, does the board have the flashing status light? (meaning everything is running) what does the light do during boot up?, does it accept gcode from a host connected to the configured serial port? (-1 is the usb serial port).

landodragon141, 2018-10-29:
Is it possible the issue is just with your screen? Can you connect it to your computer and monitor it with a host like repetier or pronterface?

p3p, 2018-10-29:
@keshantao There is something in the TMC2208 support causing Marlin to fail to initialise, if you do not select those drivers the firmware functions as normal with your config. It probably has something to do with the selected serial port pins but I'm not familiar enough with the drivers or the TMCStepper library to answer more thoroughly at the moment.

mr-miky, 2018-10-29:
TMCStepper library version ???
If it is not at least 0.22 it locks even before executing the setup ().
For me it was like that for almost two weeks before I discovered the catch.

gloomyandy, 2018-10-29:
Doesn't the OP say that he had base config + TC2208 + LCD working (or have I misread it)? So it may be something else?

p3p, 2018-10-29:
@mr-miky I was testing with 0.22, or more accurately the git repository HEAD, it is stalling when it tries to initialise the software serial in setup though.

@gloomyandy In my testing it was the only thing I had to change (his driver selection) for the firmware to work properly non of the other setting are anything out of the ordinary, but it could be an interplay with something I guess.

gloomyandy, 2018-10-29:
Do you have TMC2208s @p3p ? I ask because I'm using them but in standalone mode, I like them a lot. I've not tried using them with a serial connection (mainly because I couldn't see much of a gain from doing so, but I may be wrong).

My comment above was just based on this from the OP... "then I compiled a more plain firmware with minimal change to the configuration (with only TMC2208, enabled GLCD and changed board type), and to my surprise it worked."

I did have a quick look at the pins file and it seems OK for TMC2208 on X and Y (which I think is all the OP has defined), but the current pin configuration seems to have P2_12 used as the X RX pin AND the Z TX pin. Which is probably not going to work.

keshantao, 2018-10-29:
thanks for all the replies, I have a spare sbase, let me try to config the motor to the default one sbase use and see what will happen, and BRB :-)

keshantao, 2018-10-29:
> TMCStepper library version ???
> If it is not at least 0.22 it locks even before executing the setup ().
> For me it was like that for almost two weeks before I discovered the catch.

when it search for dependency, it says TMCStepper 0.2.2 #cefefbe

gloomyandy, 2018-10-29:
Do you have the TMC2208 serial interfaces actually connected to the J8 header on the SBase? Do you intend to use the serial interface? If not you may want to just use them in standalone mode.

keshantao, 2018-10-29:
actually I just use it plug and play, replace the on board 8825 for a quieter printer , so far I don't have the intention to play with serial control and it's not connected to J8.

gloomyandy, 2018-10-29:
In which case you have the wrong configuration. You need to use TMC2208_STANDALONE instead of TMC2208 when you define the driver types in use...
#define X_DRIVER_TYPE  TMC2208_STANDALONE
#define Y_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_DRIVER_TYPE  DRV8825

keshantao, 2018-10-30:
> In which case you have the wrong configuration. You need to use TMC2208_STANDALONE instead of TMC2208 when you define the driver types in use...
> #define X_DRIVER_TYPE TMC2208_STANDALONE
> #define Y_DRIVER_TYPE TMC2208_STANDALONE
> #define Z_DRIVER_TYPE DRV8825

you are absolutely right.  Everything works now after the change.
I missed the TMC2208_STANDALONE option in the configuration description.  Thanks alot, problem solved, need to make some minor adjustment to X direction and Y end switch.

closed 2018-10-30 by keshantao