MarlinFirmware/Marlin #12264: Clarification required on pins_STM32F4.h

vivian-ng, 2018-10-29:
Can anyone clarify what the following pins are for in pins_STM32F4.h?
#define SDSS               PE7
#define SS_PIN             PE7

I am trying to find if there are pins for the LCD12864 defined in the pins file, and it seems to be missing
SD CSEL
SD DET
RESET
and I am trying to see if these are related.

Also, I was trying to do a test compile of the 2.0 branch for STM32F4 but had errors related to ST7920 delays. It seems that ultralcd_st7920_u8glib_rrd_AVR.cpp does not contain "#elif" checks for MB(STM32F4), which is causing a "No valid condition for delays in 'ultralcd_st7920_u8glib_rrd_AVR.h'" to be thrown during compile.

Workaround was to edit ultralcd_st7920_u8glib_rrd_AVR.cpp to add in a blank #elif check for MB(STM32F4) since the delays are defined in pins_STM32F4.h

#if F_CPU >= 20000000
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(50)
#elif MB(3DRAG) || MB(K8200) || MB(K8400) || MB(SILVER_GATE)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(188)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
#elif MB(MINIRAMBO) || MB(EINSY_RAMBO) || MB(EINSY_RETRO)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(250)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
#elif MB(RAMBO)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
#elif MB(BQ_ZUM_MEGA_3D)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(189)
#elif MB(REMRAM_V1)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
#elif MB(ARMED)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
#elif MB(STM32F4)
#elif F_CPU == 16000000
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(63)
#else
  #error "No valid condition for delays in 'ultralcd_st7920_u8glib_rrd_AVR.h'"
#endif

But this is just a hack to make it compile, hopefully someone can implement a proper fix.

ktand, 2018-10-29:
Regarding the compilation error. Get the latest source from 2.0x. This was
fixed a day ago.

vivian-ng, 2018-10-29:
Thank you! Closing this issue then. I guess I was a bit slow in raising this issue, actually did the compile a few days ago and was discussing it over the Facebook page, but only got down to putting in this ticket yesterday.

As for the LCD12864 pins, I think I have got all of them, with SD_CSEL being SDSS in the pins_STM32F4.h file. The other pins were there, I just needed to open my eyes. *grin*

Closed, compile issue resolved by [12243](https://github.com/MarlinFirmware/Marlin/pull/12243)

closed 2018-10-29 by vivian-ng