MarlinFirmware/Marlin #12265: [bugfix-2.0.x] PID Autotune failed! timeout

madmax2, 2018-10-29:
### Description

While trying to PID tune my bed with the command 
M303 E-1 S100

I get error
PID Autotune failed! timeout

### Steps to Reproduce

1. User terminal in octopi, send gcode = M303 E-1 S100
2. Wait for it to auto PID tune
3. Wait some time and the error message occurs

**Expected behavior:** [What you expect to happen]
Get the PID tune results
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2524514/Marlin.zip)

**Actual behavior:** [What actually happens]
Error message:

PID Autotune failed! timeout

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

AnHardt, 2018-10-29:
How long does it take to reach that temperature without PIDTEMPBED?
Is the wanted temperature reachable at all? Even with the part cooling fan blowing on the bed?

AnHardt, 2018-10-29:
Likely your bed heater ist just not strong enough to reach that temperature in time.

```cpp
      // Timeout after MAX_CYCLE_TIME_PID_AUTOTUNE minutes since the last undershoot/overshoot cycle
      #ifndef MAX_CYCLE_TIME_PID_AUTOTUNE
        #define MAX_CYCLE_TIME_PID_AUTOTUNE 20L
      #endif
      if (((ms - t1) + (ms - t2)) > (MAX_CYCLE_TIME_PID_AUTOTUNE * 60L * 1000L)) {
        SERIAL_PROTOCOLLNPGM(MSG_PID_TIMEOUT);
        break;
      }
```
If the cycle time is above 20 minutes PID does not make sense. Just turn the heater on, and it'll do its best to reach the temperature, but when, or if it can hold that temperature with slightly more power consuming conditions, is completely open.

AnHardt, 2018-10-29:
There are lots of bed heaters out there not being able to reach 100°C. Isolating may help a bit.

madmax2, 2018-10-29:
**How long does it take to reach that temperature without PIDTEMPBED?**
I don't remember since I forgot if I timed it or not.
I can test it again to see how long.

Is the wanted temperature reachable at all? Even with the part cooling fan blowing on the bed? When the window is open?
Yes, my bed can reach 180 from the spec sheet.
I can also manually get it to reach 112 by setting it as the target temp.

Cooling/hotend fan is not blowing on the bed.
It is turned off.

So what can I do test/isolate/fix this?

BTW I also get a bed fail error, if I do a PID tune at 110

AnHardt, 2018-10-29:
A, room temperature to 110°C, temperature plot from RH(prefered with the heater graph on) or OctoPrint will tell more than thousand words.

madmax2, 2018-10-29:
so you want a screenshot of the octoprint temp graph?
or how to get the info you want?

AnHardt, 2018-10-30:
Would like to cite Tim "the Toolman" Taylor.

As a workaround:
Try to start M303 at a much higher start temperature - like 90°C.
or
Add
````cpp
#define MAX_CYCLE_TIME_PID_AUTOTUNE 40L
```
to your `Configuration.h` to give the process more time.

Be prepared to relax the 
```
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 20    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 2 // Degrees Celsius

  /**
   * As described above, except for the bed (M140/M190/M303).
   */
  #define WATCH_BED_TEMP_PERIOD 60                // Seconds
  #define WATCH_BED_TEMP_INCREASE 2               // Degrees Celsius
#endif
```
parameters.

madmax2, 2018-10-30:
@AnHardt 
So I need manually heat up the bed to 90, then start the PID Autotune?
Will this problem be fixed in future marlin 2.0.x update or is it not fixable?
Or is the fix, is that I need to just add that #define each time I compile a new firmware?

Just to understand this problem, 
in essence, marlin is expecting the bed to heat up in X amount of time
but due to it not doing so, it just gives a timeout error and stops the PID autotune?

Is this due to my psu/heatbed not being fast/powerful enough or is it due to marlin expecting an unreasonable X amount of time for it to reach 110, when it should wait more?

Can you give me some specific example values for the relax settings you posted?
Prior to your suggestion, I had already change values WATCH_BED_TEMP_PERIOD to 120 seconds.

Will this also fixed my problem with the bed failure error.
From what I read in the other issues on github, this error is due to WATCH_BED_TEMP_PERIOD  being too short, so I changed it to 120 seconds but still getting that error.

Is this is also due having thermal protection enabled so should I turn it off or increase that time period as well etc.. 
And to what is a better recommended value?

BTW do I need to create a new issue for the Bed failure error when heating up to 110 or is this error all related in this issue?

AnHardt, 2018-10-30:
In the first, rising part of your graph you see what happens when the heater is heating at full power. When extrapolating that curve you can see, the temperature will not rise infinitely but only to a point where heater energy is in balance with the losses. How big (for a temperature) the losses are can be seen in the second half of the plot (falling) where the heater energy is 0. For higher temperatures the losses are much higher then at lower temperatures.
To improve the max temperature you have to, either increase the power, or to minimize the losses.

AnHardt, 2018-10-30:
To make it clear:
"Your bed is unable to print at 110°C."
"There is no error in Marlin."

End of communication.

madmax2, 2018-10-30:
So just to clarify what you are saying..

The bed is trying to reach 110, but as it gets closer to the targeted ie >100, there is more heat loss, which makes it hard to reach that 110 value so it takes a lot longer and therefore eventually Marlin gives the timeout error due to it taking too long?

So the way to fix this is to 

increase the power = get a 24V PSU which will heat up the bed much faster
minimize the losses = put it inside an enclosure, add insulation padding under the heat bed etc?

====
Actually it can print at 110, because I have printed it at that temp before...
I am also able to increase the temp to above 110, e.g. 112 when I changed to bang bang mode.

madmax2, 2018-10-30:
This is my heatbed

https://www.amazon.com/gp/product/B00TCL9OR8?psc=1&redirect=true&ref_=oh_aui_detailpage_o00_s00&pldnSite=1&fbclid=IwAR2iBaFhn9HfGL-XdO3_kVoalpkd498B83UDXYeLhaTXtAwduDaBOnTq2O0

Specifications 
(1)Material:Aluminum Plate 
(2)Voltage:12V/24V 
(3)Max Temperature:180 Degree Celsius 

It is capable of reaching 180

AnHardt, 2018-10-30:
No! It's capable of not disintegrating blow 180°C.

kAdonis, 2018-10-30:
There is nothing wrong with Marlin and there is nothing wrong with your bed
Try insulating padding at the underside of the bed, it helps a lot.
Then start PID Autotune with the bed at 90°C and it will probably work
Keep WATCH_BED_TEMP_PERIOD as low as possible for safety reasons

madmax2, 2018-10-31:
@kAdonis Thanks for the advice, which kind of padding material do you recommend and got some link for the product?

Yes, I know this bed can do more than 110, cause I have manually pushed it to 112 in bang bang mode
and I have asked others with the same printer model, and some person has said he has printed the bed at 150..

I only noticed this problem when I use marlin 2.0x + PID, 
that is when I error the timeout error and the bed fail error.

swilkens, 2018-10-31:
> @kAdonis Thanks for the advice, which kind of padding material do you recommend and got some link for the product?

Cork mat or pipe insulating tape are decent options.

> Yes, I know this bed can do more than 110, cause I have manually pushed it to 112 in bang bang mode
> and I have asked others with the same printer model, and some person has said he has printed the bed at 150..

The listing notes this particular bed will maintain 115 C when overvolted (15V on 12V line) with no regulation - meaning constant on. If I'm not mistaken, your printer (Wanhao i3) has a 12V PSU stock? This person hitting 150 might be running it at 24V?

Looking at the specification (1.4 ohm on 12V rail, 5 ohm on 24V rail) we can see the bed pulls slightly more power on the 24V rail. So that might explain it.

- 12V -> P = ( 12 * 12 ) / 1.4 = 103 W

- 24V -> P = ( 24 * 24 ) / 5.0 = 115 W

> I only noticed this problem when I use marlin 2.0x + PID,
> that is when I get the timeout error and the bed fail error.

Is the time to heat up the bed in a previous version of marlin (say, 1.1.9) significantly different? I wouldn't expect that to be the case, but just to check.

From the specs, it seems likely that your bed is reaching the temperature limit for your situation - as @AnHardt and @kAdonis suggested. That's not bad, that's just the design.

Out of curiosity, do you actually need the bed to be at these high temperatures for your prints? If not, you can PID tune at a lower temperature that is within your heat bed's capabilities.

madmax2, 2018-10-31:
@swilkens 
**Is the time to heat up the bed in a previous version of marlin (say, 1.1.9) significantly different? I wouldn't expect that to be the case, but just to check.**

I haven't tested it on 1.1.9
I was just saying that when I am on marlin 2.0x + do an autotune PID 
I get the timeout error..

When I was on smoothieware + bang mode 
I can reach 112 manually without getting a bed fail error..
[image]

Can you explain why this is possible?

Yet when I am on marlin and using PID, if I print at 110, I get a bed fail error.
[image]

**Out of curiosity, do you actually need the bed to be at these high temperatures for your prints? If not, you can PID tune at a lower temperature that is within your heat bed's capabilities.**

The reason I wanted to PID tune at 110 and print at 110 for bed
is because that is what filament brand (esun) ABS recommends I print at 
and what google search suggest is the temp to print for ABS.

madmax2, 2018-10-31:
@swilkens

New graph

Smoothieware + bang bang mode (with hysteresis = 1)
Target temp = 116

Note = As you can see this bed has no problem going past 110 temp, and in this case it went to 115.4 (due to the hysteresis =1,  if I set target as 117, then it would go past 116..

[image]

115 and no bed fail error
[image]

swilkens, 2018-10-31:
> Note = As you can see this bed has no problem going past 110 temp, and in this case it went to 115.4 (due to the hysteresis =1, if I set target as 117, then it would go past 116..

Yes, but how fast? The Marlin PID tuning sequence will error out if the heating phase takes too long limited to 20 minutes by default. It seems you are fairly close to that - unless that time scale is in seconds?

gloomyandy, 2018-10-31:
If you are happy using bang/bang mode on Smoothieware, why not just use the same on Marlin? My understanding is that a PID controller will give you better control, but you probably can't operate it as close to the limit as you can with bang/bang. That seems very much like the issues you are having.

madmax2, 2018-11-01:
@gloomyandy Yes, using bang bang mode is an option which I know that I can use.

But when I switched over to PID mode (for more accuracy)..
I encountered this problem..
I still prefer using PID over bang bang mode because it is more recommended and accurate.

I am now just trying find out and solve why when I print at 110 it results in bed failure error.
And when I do a PID tune at 110 it results in a timeout error.
Yet when I use smoothieware + bang bang mode it is capable of going above 110.

As you can see from my graphs and LCD screen
this bed is definitely capable of going/printing at 116 and possibly more...

So I am just trying to solve this problem..

From what @swilkens has said, this is likely to be marlin's default 20 minutes timeout is too short for my printer.

It would be good if we can change the default timeout  for PID tune and printing at 110 to a value that is more suitable for my brand of printer..so that I don't get these error messages.

gloomyandy, 2018-11-01:
As I said above I don't think you are going to get good PID results and that bang/bang will probably provide better control for you. The issue is that in order to function well a PID controller needs a reasonably fast response (sometimes called good control authority) which for a heated bed means that it probably needs the system to be slightly over powered so that the output reacts well to input changes. By trying to operate the bed this close to the maximum temperature it can reach you are denying the PID controller one of the fundamentals that it requires. Even if you do manage to adjust the various PID constraints to get it to work it will probably be no better than what you can achieve with bang/bang. If you really care about this spend some time reading up on control theory, you will find that there is a lot of discussion about PID v bang/bang and you will see that that in some circumstances (like a very slowly responding control target like this), bang/bang is often the mechanism of choice. You will also notice that the default bed control mechanism used by Marlin is bang/bang, this is not just a coincidence.

boelle, 2019-02-21:
@madmax2 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead