MarlinFirmware/Marlin #12266: AUTO_POWER_CONTROL and homing/ moving axes issue

8Roman8, 2018-10-29:
### Description
Marlin bugfix 2.0 downloaded 28 Oct. 2018

maybe the issues depends on each other...

Issue 1:
if I start a printjob from octoprint, PSU turns automatically on,  bed and the Extruder start to heat,
after that Marlin recieve G28 to home all exes but nothing happens (axes dosn't move) and the Printer resets itselv after a few seconds (message on LCD to recet the Printer manually).

If bed and the nozzle is cold and I turn on the PSU with M80 or the LCD manually and heat then bed or nozzle I can send G28 or any G1 and the Printer do its Job.

i think that the Motor Drivers have to be initialized or so after the PSU goes on.

The same Thing happens when i turn on my printer (5V PSU only for Rumba board and raspberry),
I goes in the LCD menue to home or move an axis, turn the knob for example x-axis 100mm, the PSU goes automatically on but the axis doesnt move,
but if I activate the PSU manually bevore all is working fine. 

Issue 2:
after the Manual reset, the bed and the nozzle is still hot, 
so the Extruder fan should turn on (and the PSU),
in the LCD menue I can only "turn off PSU" that means the Firmware thinks that the PSU is already on.
if I send M80 manually nothing happens, only when I send M81 the PSU goes on, the same happens if I click "turn off PSU" on LCD.

I'm using TMC2130 on a Rumba board (8bit) board with x, y, z1, z2, e0, e1,
dual z-endstop on zmax and a fix mounted probe on z-min, cartesian Printer.

i will post my cfg files later.

ejtagle, 2018-10-29:
This sounds like a problem of the PSU.
Marlin turns on the PSU, waits 100milliseconds and configures the TMC steppers.

If your PSU takes longer than that time to stabilize, steppers will not be configured properly.

Check ```src/feature/power.cpp```

At function ```Power::power_on()``` there is a ```delay(100)``` . Try to increase the value to 1000 and see what happens...

gloomyandy, 2018-10-29:
With TMC2208 steppers at power up the there needs to be a VM (motor supply) present, I'm not sure if a software reset will clear any power up sequence problems. To avoid possible issues you can wire a diode to ensure correct power up sequence. See:
https://learn.watterott.com/silentstepstick/pinconfig/#silentstepsticks-with-variable-3-5v-logic-voltage
Another option is to use the stepstick protector board.

closed 2018-11-02 by thinkyhead