MarlinFirmware/Marlin #12268: [2.0.x] Status message cut-off vertically

swilkens, 2018-10-29:
### Description

The status message appears to be positioned too low on the LCD screen, causing a cut-off. 

To a layman's eye, it seems there is enough separation between the progress bar and the status message to move the status message up and prevent this.

#### Additional Information

- Running current bugfix-2.0.x (up to https://github.com/MarlinFirmware/Marlin/tree/5ead0269676b80292a360e5b132204ebecdc40a5)

 - Ender 3 Pro config

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2525922/Configuration_adv.zip)

gloomyandy, 2018-10-29:
See this PR https://github.com/MarlinFirmware/Marlin/pull/12224

swilkens, 2018-10-29:
Thanks, I'll keep an eye on it!

closed 2018-10-29 by swilkens

thinkyhead, 2018-10-31:
Patch merged.