MarlinFirmware/Marlin #12269: 2.0 "bugfix"  SPI dedicated for LCD using U8G_HAL_LINKS

mr-miky, 2018-10-29:
After several attempts I managed to make the ST7920 display work connected to a dedicated hardware SPI (STM32F407 has 3 DMA harware SPI).
To use the dedicated SPI I defined U8G_HAL_LINKS and implemented the various HAL functions that are needed.

Already on old **ultralcd_impl_DOGM.h** I noticed that, in my opinion, something was missing that allow to use U8GLIB_ST7920_128X64_4X_HAL instead of U8GLIB_ST7920_128X64_RRD and I had solved it like this:

```
// LCD selection
#if ENABLED(REPRAPWORLD_GRAPHICAL_LCD)
  #if DISABLED(SDSUPPORT) && (LCD_PINS_D4 == SCK_PIN) && (LCD_PINS_ENABLE == MOSI_PIN)
    U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_RS); // 2 stripes, HW SPI (shared with SD card)
  #else
    U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS); // Original u8glib device. 2 stripes, SW SPI
  #endif

#elif ENABLED(U8GLIB_ST7920)
  // RepRap Discount Full Graphics Smart Controller
  #if DISABLED(SDSUPPORT) && (LCD_PINS_D4 == SCK_PIN) && (LCD_PINS_ENABLE == MOSI_PIN)
    U8GLIB_ST7920_128X64_4X_HAL u8g(LCD_PINS_RS); // 2 stripes, HW SPI (shared with SD card, on AVR does not use standard LCD adapter)
  #else
    #ifndef U8G_HAL_LINKS
      //U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS);   // Original u8glib device. 2 stripes, SW SPI
     	U8GLIB_ST7920_128X64_RRD u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS);  // Number of stripes can be adjusted in ultralcd_st7920_u8glib_rrd.h with PAGE_HEIGHT
                                                                				  // AVR version ignores these pin settings
                                                                         	 	  // HAL version uses these pin settings
    #else
      U8GLIB_ST7920_128X64_4X_HAL u8g(LCD_PINS_RS);
    #endif
  #endif
```
Now the problem has moved on **ultralcd.h** and even here I had to change a bit ...

```
  // LCD selection
  #if ENABLED(REPRAPWORLD_GRAPHICAL_LCD)
    #define U8G_CLASS U8GLIB_ST7920_128X64_4X
    #if DISABLED(SDSUPPORT) && (LCD_PINS_D4 == SCK_PIN) && (LCD_PINS_ENABLE == MOSI_PIN)
      #define U8G_PARAM LCD_PINS_RS
    #else
      #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS
    #endif

  #elif ENABLED(U8GLIB_ST7920)
    // RepRap Discount Full Graphics Smart Controller
    #if DISABLED(SDSUPPORT) && (LCD_PINS_D4 == SCK_PIN) && (LCD_PINS_ENABLE == MOSI_PIN)
      #define U8G_CLASS U8GLIB_ST7920_128X64_4X_HAL
      #define U8G_PARAM LCD_PINS_RS // 2 stripes, HW SPI (shared with SD card, on AVR does not use standard LCD adapter)
    #else
      #ifdef U8G_HAL_LINKS
        #define U8G_CLASS U8GLIB_ST7920_128X64_4X_HAL
        #define U8G_PARAM LCD_PINS_RS
      #else
        //#define U8G_CLASS U8GLIB_ST7920_128X64_4X
        //#define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS     // Original u8glib device. 2 stripes, SW SPI
        #define U8G_CLASS U8GLIB_ST7920_128X64_RRD
        #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS       // Number of stripes can be adjusted in ultralcd_st7920_u8glib_rrd.h with PAGE_HEIGHT
                                                                          // AVR version ignores these pin settings
                                                                          // HAL version uses these pin settings
      #endif
   #endif
```
Maybe it's not the best solution but it's the only solution I've found ..

thinkyhead, 2018-11-04:
Seems that this is all that is needed:
[image]

Are you recommending this as a universal and permanent alteration?

boelle, 2019-02-21:
@mr-miky what thinky said ^^^^

closed 2019-05-16 by thinkyhead