MarlinFirmware/Marlin #12270: [bugfix-2.0.x] DUE + RAMPS-FD v2.1a eeprom error:-

bruce356, 2018-10-30:
### Description
Marlin-bugfix-2.0.x_71e19ba_19-10-2018
When connecting with Pronterface through Programming port:-
receive the following information;-
Connecting...
start
Printer is now online.
echo:PowerUp
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Oct 30 2018
echo: Free Memory: 83308  PlannerBufferBytes: 1664
Error:EEPROM datasize error.
[ERROR] Error:EEPROM datasize error.

echo:Index: 537 Size: 633
echo:Hardcoded Default Settings Loaded
echo:SD card ok

### Steps to Reproduce
Load marlin firmware configured to use eeprom and connect to host Pronterface

#### Additional Information
If attempting M502 M500 M501 receive following errors:-
>>> M502
SENDING:M502
echo:Hardcoded Default Settings Loaded

>>> M500
SENDING:M500
echo:Settings Stored (621 bytes; crc 62659)
Error:EEPROM datasize error.
[ERROR] Error:EEPROM datasize error.

>>> M501
SENDING:M501
Error:EEPROM datasize error.
[ERROR] Error:EEPROM datasize error.

echo:Index: 621 Size: 633
echo:Hardcoded Default Settings Loaded

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2529404/Marlin.zip)
Regards - bruce

hectori4502, 2018-10-30:
test with another sd.

gloomyandy, 2018-10-30:
How recent is that 2.0.x? There have been a number of reports of similar problems and various fixes over the last week or so. You should probably retest with the latest version.

MrAlvin, 2018-10-30:
What is written on the I2C EEPROM chip on the RAMPS-FD shield? 
Or in other words, how many bytes can be stored in the chip? 

What does your pins_RAMPS_FD_V2.h file say about the size of the eeprom chip? 
Look for something like:  #define E2END 0xFFFF // 64K in a 24C512
or: #define E2END 0x1FFF // I2C EEPROM with 8K of space

rafaljot, 2018-10-30:
AT24C512 , 512Kbits (65KB)

bruce356, 2018-10-31:
Hi, Guys thanks for the replies, @hectori4502 changing SD card or turning of the facility made no difference.

@gloomyandy I was using Marlin-bugfix-2.0.x_71e19ba_19-10-2018 and updated to Todays version and the problem has been resolved - thanks

@MrAlvin tried changing to "#define E2END 0x1FFF // I2C EEPROM with 8K of space" before updating to latest version and made no difference - thanks

@rafaljot you are right its working fine with latest version, what a difference 11days make.

Regards - bruce

closed 2018-10-31 by bruce356

hectori4502, 2018-10-31:
this it´s my last stable firmware (after October 4 any compile firmware get too many warnigs) MKS Sbase Delta.
[Marlin-bugfix-2.0.xOct4.zip](https://github.com/MarlinFirmware/Marlin/files/2536353/Marlin-bugfix-2.0.xOct4.zip)

thinkyhead, 2018-10-31:
Warnings aren't too concerning. There are several that we can't do anything about.